MajicDesigns/MD_Parola #1: ESP8266: error: redeclaration of 'CLOSING'

biohazardxxx, 2016-11-25:
Hi,

when I included Parola and this library into my project where I need ofcourse WiFi connectivity the compile fails. With error: redeclaration of 'CLOSING'.
Without Wifi it works ok _(while most of the out effects don't work, but ok for me)_.

> 
> ...\arduino-1.6.9\portable\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:28:0,
> 
> from ...\arduino-1.6.9\portable\sketchbook\MySketch\ESP8266\My\MAX7219_WiFiMqtt_Display\MAX7219_WiFiMqtt_Display.ino:3:
> 
> ...\arduino-1.6.9\portable\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/include/wl_definitions.h:81:17: error: redeclaration of 'CLOSING'
> 
>    CLOSING     = 8,
>                  ^
> 
> In file included from ...\arduino-1.6.9\portable\sketchbook\MySketch\ESP8266\My\MAX7219_WiFiMqtt_Display\MAX7219_WiFiMqtt_Display.ino:1:0:
> 
> ...\arduino-1.6.9\portable\sketchbook\libraries\Parola\src/MD_Parola.h:305:3: note: previous declaration 'textEffect_t CLOSING'
> 
>    CLOSING,    ///< Appear and disappear from the ends of the display, towards the middle
> 

Is that something that can be fixed? Or do you have a suggestion for a workaround?

BR

MajicDesigns, 2016-11-25:
Try the latest repo code as this has been fixed.

biohazardxxx, 2016-11-25:
updated both Parola and MAX72XX.
and now I get this (error from Parola_HelloWorld example):

> 
> c:\Users\bla\AppData\Local\Temp\arduino_modified_sketch_92268\Parola_HelloWorld.ino: In function 'void setup()':
> 
> Parola_HelloWorld:26: error: 'CENTER' was not declared in this scope
> 
>    P.displayText("Hello", CENTER, 0, 0, PRINT, NO_EFFECT);
> 
>                           ^
> 
> Parola_HelloWorld:26: error: 'PRINT' was not declared in this scope
> 
>    P.displayText("Hello", CENTER, 0, 0, PRINT, NO_EFFECT);
> 
>                                         ^
> 
> Parola_HelloWorld:26: error: 'NO_EFFECT' was not declared in this scope
> 
>    P.displayText("Hello", CENTER, 0, 0, PRINT, NO_EFFECT);
> 
>                                                ^
>

MajicDesigns, 2016-11-25:
All the effect types now have PA_ as prefix. I thought all the examples were already changed over. Is this an old example you had from before?

biohazardxxx, 2016-11-25:
yes true I did not load up a fresh one. works just wonderfull now. Thanks.

closed 2016-11-25 by biohazardxxx