MajicDesigns/MD_Parola #2: Add to Arduino IDE library manager

bperrybap, 2017-01-01:
It would be great to add this library to the Arduino IDE library manager.
That would allow users to quickly and easily install it directly from the Arduino IDE GUI in just seconds.
It keeps the less technical users from doing things like installing it incorrectly and allows them to easily updated it when updates come out.

It isn't difficult to do, (you have to get the Arduino team to add it - but that is a one time thing)
Then after that, when you have a new release all you have to do is create a tag and the library manager will discover the new release in about an hour.
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

BTW, it would also be helpful to add some installation instructions to the readme.

MajicDesigns, 2017-01-01:
One of the reasons for moving the repo to GitHub is to include it in the IDE library manager, so that is in the plans for the near future.

Install instructions are a good idea and I'll do something, but the biggest issue I find is that a substantial minority do not read any documentation - they just 'load and go'. Not sure how o get the message to that subset. Automatic installation from the IDE is not going to help this as you don't even get to see the readme unless you make an effort, which they don't.

bperrybap, 2017-01-01:
I have the same issue with some of my libraries.
But at least with install instructions in the readme for the repo you can provide a link to them that gets them to the repo where they can read how to install it using the library manager.
There are still quite a few Arduino users that have no idea that the library manger even exists.
It can also help for those that might stumble across the repo using a google search.

Also one thing that can help is if the example sketches have a  web link in them that points to repo and potentially a place to find any documentation.
That way once the library is installed, there are always links being put in front of them of where to find additional information. And with newer IDEs they can click on the links and bring up a browser directly from the IDE.

Recently, I've starting taking bit harder approach and simply not helping those very much that either won't or refuse to read the documentation. I'll politely point them to the documentation.
Either they finally read it or give up. Not sure but either way, the amount of silly follow questions has dramatically reduced.

--- bill

MajicDesigns, 2017-01-05:
Libraries have been added to the download manager. Additional comments have been put in the MD_Parola example files. Thanks for the suggestion.

closed 2017-01-05 by MajicDesigns

bperrybap, 2017-01-05:
One thing to consider with direct links to documentation in the examples, is that examples may end up pointing to updated documentation that is not applicable to the version of the library installed.
I have a similar issue in my libraries.
I'm considering having a different documentation directory for each library version and then having a fixup script that goes in and patches all the examples to point to the correct location for that version of the library.
In my case I already have some fixup scripts that patch things like library version information in various files and this would be just another fixup.

There is no easy solution for handling per revision documentation.