MajicDesigns/MD_Parola #4: A way to get a pointer to the Hardware library object

pockemul, 2017-01-15:
I need to shutdown the display when i put the ESP8266 in deepsleep.
But there is no way to retrieve the _D object.
Would it be the good way to add in MD_Parola.h:
`MD_MAX72XX* getMAX() { return &_D;}`

and to put in my code:
`P.getMAX()->control(MD_MAX72XX::SHUTDOWN, true)`

MajicDesigns, 2017-01-15:
I would prefer not to provide access to _D as tech support is hard enough as it is. The idea is valid though and I can implement a method to shut down the display, same as you can do set intensity.

pockemul, 2017-01-15:
I understand your concerns about tech support.
I think you're right keeping access to _D not possible. If a user has enough skills to need Hardware object access, he can add the function as i did.
A function to shutdown and wake up the display in MD_Parola would be great.

Thank you for your amazing job

MajicDesigns, 2017-01-16:
Added displayShutdown() and zoneShutdown() methods. Updated doc files. Available in the Repo.

closed 2017-01-16 by MajicDesigns