MajicDesigns/MD_Parola #6: Custom font causes compiler error (when used with aREST)

marcfon, 2017-05-14:
I'm trying to combine MD_Parola with aREST but it's throwing a compiler error at me. This error is only generated when I enable a custom font. It has to do with the **PROGMEM** identifier but my arduino knowledge it too limited to understand/fix the issue here. It could very well be that the issue is not caused MD_Parola but by the aREST lib in that case I apologise for posting it here.

This is the full error I'm getting.
`
src/main.cpp:23:32: error: fontSysFixedSingle causes a section type conflict with __c
MD_MAX72XX::fontType_t PROGMEM fontSysFixedSingle[] = {};
^
In file included from /Users/marcfonteijn/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:243:0,
from src/main.cpp:1:
/Users/marcfonteijn/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/pgmspace.h:21:51: note: '__c' was declared he
re
#define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
^
/Users/marcfonteijn/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:38:76: note: in definition of macro
 'FPSTR'
#define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
^
/Users/marcfonteijn/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:39:34: note: in expansion of macro 
'PSTR'
#define F(string_literal) (FPSTR(PSTR(string_literal)))
^
lib/aREST_ID429/aREST.h:1456:15: note: in expansion of macro 'F'
addToBuffer(F("\", \"connected\": true}\r\n"));
^
`

The code below isn't doing anything useful but it's the minimum needed to reproduce the compiler error.

`
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <aREST.h>

// matrix display
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>

//------- aREST Settings ------

aREST rest = aREST();

//------- Matrix Settings ------
#define MAX_DEVICES 4
#define CLK_PIN   D5
#define DATA_PIN  D7
#define CS_PIN    D8

// Hardware SPI connection
MD_Parola P = MD_Parola(CS_PIN, MAX_DEVICES);

MD_MAX72XX::fontType_t PROGMEM fontSysFixedSingle[] = {};

void setup() {
  Serial.begin(115200);
  Serial.println("aRestTest");

  P.begin();
  P.setFont(fontSysFixedSingle);
}

void loop() {
    // Do something useful here
}
`

MajicDesigns, 2017-05-14:
What happens if you comment out the references to aREST (what is that?) and try and compile?

marcfon, 2017-05-14:
Yeah, code compiles just fine without aREST. It's somehow the combination of aREST and custom fonts. 

There are 3 ways to make this code compile without errors.
1. remove aREST (https://github.com/marcoschwartz/aREST)
2. remove the PROGMEM flag from the custom font declaration
3. comment out P.setFont(fontSysFixedSingle);

Just to be sure I've downloaded the most recent libraries again but the compiler error still remains.

MajicDesigns, 2017-05-15:
Installed the aREST library and tried to compile your code (without WiFi library include). Compiled with quite a few warnings from the aRest library, but it completed ok. My setup is Arduino IDE 1.8.2, Windows 10, Arduino Uno.
`Sketch uses 17760 bytes (55%) of program storage space. Maximum is 32256 bytes.
Global variables use 261 bytes (12%) of dynamic memory, leaving 1787 bytes for local variables. Maximum is 2048 bytes.`

marcfon, 2017-05-16:
So I've noticed that it has something to do with the fact that I'm compiling for a ESP8266 board (wemos d1 mini). When I set the board to an Arduino it compiles just fine like you said.

I've tried compiling it using Arduino 1.6.12, 1.8.2 and platformio 2.0.0-beta3 on OSX 10.11.6. It keeps returning this compiler error: `fontSysFixedSingle causes a section type conflict with __c`

When compiling for the ESP8266 I must include the **<ESP8266WiFi.h>** library otherwise aRest starts complaining that WifiClient is missing.

MajicDesigns, 2017-05-16:
You are probably going to need to dig around the libraries and/or google for any issues with PROGMEM with the Wemos.

marcfon, 2017-05-16:
Thanks for the right pointer @MajicDesigns! I've got it to compile based on [this post](https://halfbyteblog.wordpress.com/2015/03/15/progmem-issues-with-arduino-1-6-x-and-how-to-fix-them/). To be honest I still don't fully understand what is causing this issue but it's good enough for me now. Now I'll try to assemble a fully functional piece of code and see if it runs as expected.

I had to add this code block.
```C++
#ifdef PROGMEM
#undef PROGMEM
#define PROGMEM __attribute__((section(".progmem.data")))
#endif
```

closed 2017-05-16 by MajicDesigns

dan2600, 2018-03-26:
sorry to dig up an old issue but I'm having the same problem with a custom font on the ESP8266 

const uint8_t DejaVu_Sans_10[] PROGMEM = { //font data};

I tried marcfons fix

const uint8_t DejaVu_Sans_10[] __attribute__((section(".progmem.data")))= { //font data};

but I get the same error.

MajicDesigns, 2018-03-26:
Issue #1 (closed) here seems to indicate that it will compile ok with the ESP8266. PROGMEM is a defined keyword in the ESP8266 header files.

dan2600, 2018-03-26:
I'm very new to C++ and mostly program in higher level languages so I'm not sure how good my fix was, but what I did was save my custom font into a separate namespace and then just reference it when i called it into the library .

I then got the same error based on the default font being used by the library (ARIAL_10 I think). I'm too much of a novice to modify a library, but I just modified fonts.h and have the default font not save in PROGMEM (SRAM instead). Not efficient, but the best I can figure out right now.

It's working though, this is a great library!

MajicDesigns, 2018-03-26:
Ok. If this is an ongoing problem, please raise a new issue in future so I can track it properly. Thanks.