MajicDesigns/MD_Parola #7: Can you precize documentation

mike1796, 2017-07-07:
Hi, 

I'm working on a Daft Punk Thomas helmet and I'm trying to run your code on a 8x40 LEds matrix using an Arduino Uno and 5 MAX7221 chips. I'd like to execute your code to display text on my matrix but I'm a complete newbie and it won't stop failing to compile with Arduino Software (1.8.3).
Could you help me or specify this in your documentation.

Thanks for your time.

MajicDesigns, 2017-07-07:
What is the error message for the fail? There is a Daft Punk example in the MD_MAX72xx library examples - is that what you are trying to run?

You need to be more specific before I can help you.

MajicDesigns, 2017-07-10:
This issue will be close in 3 days.

mike1796, 2017-07-13:
Hi MajicDesigns, thanks for the consideration. The error message I got was something like "Can't find MD_Parola_H library", but as I said I'm completely new to this stuff and there's something I might be doing completely wrong. Didn't even knew there was a Daft Punk example in your libraries! Any chance you can give a step by step note on how to correctly compile that program on my Arduino? 

Thanks again!

MajicDesigns, 2017-07-13:
You need to install the libraries MD_Parola and MD_MAX72xx in a "libraries" subfolder of your Sketchbook folder. One easy way to do this is to find the libraries using the Library Manager menu option in the IDE and add them in from there. For details on how to compile Arduino programs - please use Google and find other tutorials on how this is done. There is nothing special about these examples programs once you understand the basics of Arduino libraries and compiling.

Please also take time to read the documentation for the libraries. There is a lot of information there that you will be able to use and avoid frustration.

mike1796, 2017-07-21:
Hi MajicDesigns, We just understand how everything works. Thanks for your amazing help ! it's finally working

closed 2017-07-21 by mike1796