MajicDesigns/MD_Parola #9: for 1088AS 8x8 displays

ieee488, 2017-08-11:
Not verified.
1st time effort.
I hope it is correct.

[Parola_v1-2.zip](https://github.com/MajicDesigns/MD_Parola/files/1219221/Parola_v1-2.zip)

knopserl, 2019-08-30:
I have a 4x8.8 1088AS LED block and it shows the text horizontaly mirrored.
I use the latest 3.1 version (with an ESP8266, NodeMCU) and have tried several HW types but the bext is: #define HARDWARE_TYPE MD_MAX72XX::PAROLA_HW
but it is still horizontally mirrored.
Any ideas?

knopserl, 2019-08-30:
I have found a setting for this display which works:
#define MAX_DEVICES 4 // Define the number of displays connected.
#define HARDWARE_TYPE MD_MAX72XX::ICSTATION_HW
#

Dist0rtion, 2019-09-05:
```
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 4
```
```
// SOFTWARE SPI
MD_Parola P = MD_Parola(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);
```
This is working better i think ... at least for me
Runs good on an Wemos D1 Mini

closed 2019-10-20 by MajicDesigns