MajicDesigns/MD_Parola #10: avr/io.h ESP8266

Ukhauk, 2017-09-11:
Hi downloaded via Library Manager but get error when compiling the ESP8266 avr/io.h 

Example trying to compile Parola_Scrolling_ESP8266

Whats the best way to work round the issue ?

Any advice appreciated

MajicDesigns, 2017-09-12:
Can't give advice unless I know what error you are getting.

MajicDesigns, 2017-09-12:
Can't give advice unless I know what the error is.

Ukhauk, 2017-09-20:
Thanks for your help only just noticed your reply

Arduino: 1.8.3 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

C:\Users\Documents\Arduino\libraries\MD_MAX72XX\src\MD_MAX72xx_font.cpp:24:21: fatal error: avr/io.h: No such file or directory

 #include <avr/io.h> 

                     ^

compilation terminated.

MajicDesigns, 2017-09-20:
The error is what it says it is. I am guessing that the WeMos is not avr archittecture, so that file is not appropriate. You will need to find the equivalent file for WeMos or maybe just leave it out.

Ukhauk, 2017-09-21:
Hi what I don't understand is I get the error when compiling the ESP8266 example "Parola_Scrolling_ESP8266" I have set the board type to generic ESP8266 4meg.

Is this issue just specific to my setup ? to make this library truely an IOT  its got to be driven by an ESP8266 or ESP32 to update display information.

All the other examples compile and run fine on my arduino's Its a fantastic library

Any help appreciated here is the error with generic ESP8266 board selected avr/io.h is used in the font file MD_MAX72xx_font.cpp

Arduino: 1.8.3 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 4M (1M SPIFFS), ck, Disabled, None"

Build options changed, rebuilding all
C:\Users\Documents\Arduino\libraries\MD_MAX72XX\src\MD_MAX72xx_font.cpp:24:21: fatal error: avr/io.h: No such file or directory

 #include <avr/io.h> 

                     ^

MajicDesigns, 2017-09-21:
can't see where it is included in the font cap file (check the repository...). Suggest you should download the Max72xx and Parola latest versions if you have not already done so.

MajicDesigns, 2017-09-22:
Tested compiling the example with IDE 1.8.4, selected hardware ESP8266 GEneric module. No errors reported on my system.

closed 2017-09-22 by MajicDesigns

Silberlachs, 2018-02-12:
for some reason the include with avr files does not work properly.
i am not flaming you for not posting your code as many other members <.<

try something like this
#include "C:/Program Files (x86)/Arduino/hardware/tools/avr/avr/include/avr/sfr_defs.h"

note the quotes instead the < > notation, this is for including since the old c days and means:

" xxx " ---> include from the folder the program code is in (or the path given, see above)
< xxx > ----> include from the standart library path

now you might wonder why arduino can't find the standart libraries in it's own installation directory.
the answere COULD be that they recently changed the standart include path to /documents/Arduino/libraries/

assuming you are on windows, don't know for linux and don't know anybody using mac