MajicDesigns/MD_Parola #11: Wipe does not work for one segment  (zone)

petervflocke, 2017-10-21:
Hi Marco,

testing your great library I found out that wipe (PA_WIPE) effect does not work for one segment display nor for one segment zone area. 
Environment: ESP32 + MAX7219 + USE_ICSTATION_HW

How to reproduce:
Chane the example Parola_Animation_Example like this
#define MAX_DEVICES 1
in the sCatalog table shorten the psz from "WIPE" to "W"  e.g.  { PA_WIPE, "W", 5, 1 },

The program hangs and does not leave the animation, stays there for ever.
I haven't yet troubleshoot this, juts wanted to let you know :)

...anyway great job!
P.

MajicDesigns, 2017-10-21:
Verified this behaviour only occurs with centered text. Please confirm this is the same on your setup.

MajicDesigns, 2017-10-22:
Similar behavior found for SCAN and GROW effects. Should now be fixed and uploaded to the repository.

closed 2017-10-22 by MajicDesigns

petervflocke, 2017-10-23:
> Verified this behaviour only occurs with centered text. Please confirm this is the same on your setup.
That's right - forgot to mentioned this. Thank you for posting patches.