MajicDesigns/MD_Parola #12: Parola_Double_Height_Test.ino

mr-miky, 2017-10-24:
Hi, 

I took a lot of time to understand why the various effects of the demonstration do not work and in the end I found out something was wrong here:

memcpy_P(&ci, catalog + (idxCat * sizeof(catalogItem_t)), sizeof(catalogItem_t));

I'm using 3 zones of 8 modules so 

typedef struct catalogItem_t
{
  bool fPause;
  uint8_t zFX[MAX_ZONES];
};

is 1 byte for bool and 3 bytes for zone FX.

and in fact  sizeof(catalogItem_t) return 4.

At first loop idxCat is 0 so  catalog + ( 0 * 4) so 

memcpy_P(&ci, catalog ), 4);

catalog , if I remember correctly, is 0x110A.
At second loop I would expect:

memcpy_P(&ci,catalog + (1 * 4),4)   --->>> memcpy_P(&ci,0x110A + (1 * 4),4)    that becomes  memcpy_P(&ci,0x110E,4) instead it becomes memcpy_P(&ci,0x111A,4) and each loop the flashram pointer increase of 0x10 instead of 0x04.

Obviously for first loops maybe you are lucky to read an effect that should be done 4 times later but in the later ones you read some of the code in flash (maybe the characters).

I have solved in that way:

    Serial.println("");
    Serial.print("memcpy_P() flash address = ");
    Serial.println( (unsigned long)catalog + idxCat * sizeof(catalogItem_t),HEX);
    Serial.print("memcpy_P() flash address = ");
    Serial.println( (unsigned long)(catalog + (idxCat * sizeof(catalogItem_t))),HEX);

    // copy the next catalog item into RAM
   memcpy_P(&ci, (const void *)((unsigned long)catalog + (idxCat * sizeof(catalogItem_t))),    sizeof(catalogItem_t));

serial debug result :

[Double_Height_Test]
Allocating memory: 7
Changed speed to 44
C0: M0: 876
memcpy_P() flash address = 116E
memcpy_P() flash address = 116E

Changed speed to 24
C0: M1: LuiZa
memcpy_P() flash address = 116E
memcpy_P() flash address = 116E

Changed speed to 19
C1: M0: 876
memcpy_P() flash address = 1172
memcpy_P() flash address = 117E

C1: M1: LuiZa
memcpy_P() flash address = 1172
memcpy_P() flash address = 117E

C2: M0: 876
memcpy_P() flash address = 1176
memcpy_P() flash address = 118E

C2: M1: LuiZa
memcpy_P() flash address = 1176
memcpy_P() flash address = 118E

C3: M0: 876
memcpy_P() flash address = 117A
memcpy_P() flash address = 119E

C3: M1: LuiZa 
memcpy_P() flash address = 117A
memcpy_P() flash address = 119E

C4: M0: 876
memcpy_P() flash address = 117E
memcpy_P() flash address = 11AE

MajicDesigns, 2017-10-25:
What hardware and IDE version are you using?

mr-miky, 2017-10-26:
Tested on Arduino UNO and MEGA 2560
IDE 1.6.19 
IDE 1.6.15

the same result

C0: M6: Wikiberto
memcpy_P() flash address = 116E  OK
memcpy_P() flash address = 116E  OK

C1: M0: 876
memcpy_P() flash address = 1172  OK
memcpy_P() flash address = 117E  WRONG !

C1: M1: Taz!
memcpy_P() flash address = 1172 OK
memcpy_P() flash address = 117E WRONG !

MajicDesigns, 2017-10-28:
Changed to 
memcpy_P(&ci, &catalog[idxCat],  sizeof(catalogItem_t));
seems to fix the problem.

closed 2017-10-28 by MajicDesigns