MajicDesigns/MD_Parola #14: Strange Behaviour

mr-miky, 2017-10-30:
Ciao Marco,

I have assembled a display of 3 zones of 8 displays and adapted the demos Parola_Double_Height_Test.ino and Parola_Bluetooth_Control.ino.
Apart from the previous problems that you have already fixed, another one I had noticed but attributed to a power supply problem.
Now I've found that if I turn off the DEBUG option, the display stops at startup by displaying meaningless characters but always equal to each turn on.
Thinking he had missed something I commented on all PRINTX, PRINT, #define, etc. The program compiles correctly and always behaves the same way, meaningless caracters.
I added in setup () Serial.begin (57600);
No change in behavior.
I added an additional Serial.println ("");
Miracle, the display works properly.

At this point I do not understand why this happens, I suspected the interrupts not enabled, I added interrupts (); in setup () instead of Serial.begin and Seial.print but it does not work anyway.
It seems that Serial.print initiates something that libraries do not do ....

MajicDesigns, 2017-10-30:
Do the single line display examples run eg Parola Test? What about the std 2 line examples?
Do you have external power or just usb power?

mr-miky, 2017-10-30:
Dedicated external psu for display/dots . PC USB or 2A powerbank for arduino.

In the meantime I found out if I put Serial.print ("A") in place of Serial.println ("") the problem persists but if I put Serial.print ("AA") it works correctly.
It seems that at least 2 characters must be out of the serial.

It seems all nonsense ....

MajicDesigns, 2017-10-30:
Are the power supply grounds all connected together?

mr-miky, 2017-10-30:
The grounds are obviously connected together

I've been working for almost 30 years to design / build / repair electronic devices now I'm not surprised at almost anything :-)

Maybe the bootloader leaves the microcontroller in some strange state ....

mr-miky, 2017-10-30:
Recovered bootloader, controlled 5V, and 3.3V with the oscilloscope, are corrected as level and "beautiful clean"

There is still no logical explanation of the phenomenon.

MajicDesigns, 2017-10-30:
OK. I have no idea if you have 30 years experience or you started yesterday, so the obvious questions have to be asked.

There was some discussion of this in the Parola forum thread some time back from a Russian guy who was also using the double height arrangement. I think he ended up adding a delay in startup() and it fixed the problem, but it would be worth you looking there as well.

I also asked before if you get this problem with only single height display or the other 'standard' example code.

mr-miky, 2017-10-30:
I see. They are the same things that I ask when I am trying to give answers to others in the same situation :-)

I changed Arduino Mega 2560 and things still strange.

I suspect there is something wrong with the ZONES, if I do 1 zone of 24 items instead of 3 of 8 everything starts right ( without Serial.print ) apart from the fact that the fonts are small and unreadable at a distance

MajicDesigns, 2017-10-31:
The memory for tracking the zones is allocated at the start of the object creation (in the begin() for MD_Parola class), as _Z. Maybe that memory allocation is failing or something not working right? Memory is allocated the same if it is one zone.

mr-miky, 2017-10-31:
Ciao Marco,

I do not understand anything more. Tonight I was awake up to 3am ....
I abandoned my code and used a clean demonstration to which I only added what is needed for the third zone.

[Parola_Double_Height_v1.zip](https://github.com/MajicDesigns/MD_Parola/files/1432348/Parola_Double_Height_v1.zip)

As you can see there is nothing strange or at least I do not see it.

Well if it was a problem with low memory availability it would explain why 3 7-zone zones start correctly and a 3 x 8 but not why  a 3 x 9 works correctly ?? :-(

Obviously adding DEBUG 1 to 3 x 8 code works all (debug generates a lot of Serial.print ()).
Meanwhile I tried to program the ATMEGA directly without the bootloader but it did the same anomalies.

I've read somewhere in the documentation, but I can not find out how much memory is allocated.
I have the impression that I will be forced to attack some led and check if some malloc () fails ...

Maybe it's caused by the optimization of gcc ... 

I forgot I tried with Arduino IDE 1.85 official although I normally use eclipse with the plugin Sloeber (http://eclipse.baeyens.it/)

MajicDesigns, 2017-10-31:
Memory requirements are documented in a blog article I wrote https://arduinoplusplus.wordpress.com/2017/08/27/parola-a-to-z-ram-requirements/

I don't think the bootloader is a problem. I also don't think the programming tools environment is a problem, as they all end up using gcc to compile the code.

The use of a LED to check if the allocation is a good idea.

Overnight someone else has reported a similar issue in the Parola thread in the Arduino forum http://forum.arduino.cc/index.php?topic=171056.new#new, post #1158.

I think this is starting to look like a memory issue of some sort.

mr-miky, 2017-11-01:
Meanwhile I have, even if it does not excite me, installed Atmel Studio 7 on pc with windows that almost never use, I'm a linux user :-) .
In this way I can use the Atmel ICE debugger and see step by step what happens. I hope I will not go to bed even at 3 o'clock tonight ...

mr-miky, 2017-11-01:
Hi Marco,

After a day of work I have never been able to see the problem on the debugger of atmel studio 7.
All malloc() are right and I have checked them by running the code step-by-step.
I'm getting more and more convinced that the avr-gcc toolchain does some nasty joke (as it did for demo double height).
My suspicion arises from the fact that ATMel studio uses avr-gcc 5.4 and also does not use Os but O1 optimization.
Arduino Suite 1.85 uses Gcc 4.9.2.
For today is enough ...

MajicDesigns, 2017-11-01:
Thanks for the update

mr-miky, 2017-11-02:
Grrrrr ....

Right now I'm really poisonous like a rattlesnake ...
After that series of reflections I used the Arduino 1.85 ide on windows with BOARD 1.6.19 and the result is the same I get by compiling on linux, arduino locks at the start ....
Downgraded with board 1.6.9 ..... everything works properly ..
Upgrade to board 1.6.20 ... surprise ... it does not work anymore.

I tried until 1.6.10 works, 1.6.13 sometimes starts the 1st zone  shifted right of an display element.
Anyway now I'm working with 1.6.10 without problems, maybe the gcc does not like much what you do with pointers ...

P.S.  Board 1.6.10, 1.6.11 uses gcc 4.8.1  , Board 1.6.12, 1.6.13 , 1.614 gcc 4.9.2 ,  Avr studio7 gcc 5.4

closed 2017-11-02 by mr-miky

MajicDesigns, 2017-11-02:
Interesting. I don't do anything non-standard with pointers, though. Of course, it may not be in my code that the problem occurs but somewhere that is called by my code (in the standard libraries?). Anyway, you seem to have a solution but it is very unsatisfying why this is happening.