MajicDesigns/MD_Parola #16: PlatformIO

ageir, 2018-01-12:
I'm trying to get MD_Parola to work with PlatformIO. http://platformio.org/lib/show/1389/MD_Parola
The library is available via the library manager. But the required MD_MAX72xx library is not available as far as I can see. And because of that the MD_Parola library is not working.

Any idea why it's not available as a package? Or even included in the Parola package.

Hope anyone can point me in the right direction on how to solve this. Thanks.

MajicDesigns, 2018-01-12:
MD_MAX72xx library is available at https://github.com/MajicDesigns/MD_MAX72XX, not that hard to find.

I don't officially support Platform I/O, so no idea how the library ended up in their catalog.

MajicDesigns, 2018-01-12:
Closed accidentally.

ageir, 2018-01-13:
Ah. I assumed it was you that uploaded it. Strange. Maybe you can send an email to the person that uploaded it. The email is listed on the library page.
Don't really understand why that person didn't upload the core library. There are a lot of versions of the parola library on there. Just wierd.

I got it working by downloading the library separately. No issues at all.
Would be great to have it in the PlatformIO package management though. Makes it real easy to upgrade and change versions. Hope you'll support it in the future.

MajicDesigns, 2018-01-13:
I'll look into it.

closed 2018-01-13 by MajicDesigns

ageir, 2018-01-13:
Great! :+1: 
By the way. Is it possible to connect 3 or 4 modules by height? Would like to make a 8x4 matrix.
I saw it mentioned in the blog that you can have dual height, but no mention that it isn't possible with more.

MajicDesigns, 2018-01-13:
You can connect modules by height but you need to make sure the font definition creates characters that span across as many modules in height as you need. In principle using >2 height is no different from using 2. Find more info on this in my blog at https://arduinoplusplus.wordpress.com, look for articles in the "Parola A to Z Series".

ageir, 2018-01-14:
Cool! Thanks. I already read the blog post on dual height displays. And it only mentions double height.
It kind of implies that more than two in height is not supported. I suggest that you update that post and documentation to say that it supports more. I think a lot of people just skip using your library if they want to use larger displays because it's not mentioned. Maybe somewhere hidden in the manual? Haven't read everything yet. Could at least not find it easily with google.