MajicDesigns/MD_Parola #17: MQTT disconnects while displayText

debsahu, 2018-02-15:
The MQTT connection drops while using PubSub and animation using P.displayText. PubSub and MD_Parola dontplay well with each other? Using a ESP8266.

debsahu, 2018-02-15:
Thanks for the quick reply.

Here is the code: https://github.com/debsahu/Scrolling-Text-Home-Assistant/blob/master/Scrolling_Text_Home_Assistant/Scrolling_Text_Home_Assistant.ino

I agree that Animate() takes some time, but is there a way to insert "mqtt.loop(); yield();" in some way while animating? Just point me to where in library that I need to change.

MajicDesigns, 2018-02-16:
The point is that animate() by itself takes almost no time at all, so the problem is more likely somewhere else in your code.

I would expect that "client.loop(); yield();" should be performed EVERY TIME through loop(), not just as an 'else' condition. They work the same way that animate() does - they need to be called as often as possible.

debsahu, 2018-02-16:
While MQTT connection is maintained, "client.loop(); yield();" is done. If it disconnects, it tries to reconnect. This is the non-blocking MQTT implementation, even with this it fails. And I already have them outside the animation loop as well.

MajicDesigns, 2018-02-16:
I still think your structure for calling the yield() and client.loop() methods is wrong, but I am not writing your code.

Then you will just have to try different things until it works. Changing all the animations to PRINT will be a start, as these finish in 1 cycle (a few milliseconds).

debsahu, 2018-03-04:
I figured out the WiFi & MQTT dropouts. The analogread needs a delay which I did not include. This is no fault of MD_Parola, please close this (non)-issue.

MajicDesigns, 2018-03-04:
ok. Good to hear you fixed your problem.

closed 2018-03-04 by MajicDesigns