MajicDesigns/MD_Parola #20: dependancy on MD_UISwitch

pacav69, 2018-03-27:
While compiling Parola_Test_TG.ino this error occured
/Documents/Arduino/libraries/MD_Parola/examples/Parola_Test_TG/Parola_Test_TG.ino:21:25: fatal error: MD_UISwitch.h: No such file or directory
compilation terminated.

It appears that this library is dependant on 'MD_UISwitch'
can be fixed by installing MD_UISwitch using the library manager.

MajicDesigns, 2018-03-27:
Not sure how this is an issue. It clearly states at the top of the .ino file that 
"// UISwitch library can be found at https://github.com/MajicDesigns/MD_UISwitch"

closed 2018-03-27 by MajicDesigns

pacav69, 2018-03-27:
i thought you could place the reminder/warning in the readme file

MajicDesigns, 2018-03-27:
Every example uses one or more libraries and these are highlighted in the examples. The readme file is not read by most people using the library.