MajicDesigns/MD_Parola #22: P.synchZoneStart() not working properly

sankaramp, 2018-04-09:
hello,
i made a display with 16x2 (2rows and 16columns).Total 32  8x8 matrices.
i wrote a code for double height display demo...Total 32 devices taken as two zones(bottom zone and Top Zone),,,in that P.synchZoneStart(); is not working properly...When data is scrolling the two zones are not in synchonization....the bottom zone is slight faster than top zone....

please give me suggestions,
thanq,
Sankar.

MajicDesigns, 2018-04-09:
What is your matrix arrangement?
What code are you running?
HAve you tried the example code and does that work properly?

sankaramp, 2018-04-11:
here is code,,,plz have a look.,

#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
#include "Font_Data.h"

#define MAX_DEVICES 32
#define CLK_PIN   PB6
#define DATA_PIN  PB7
#define CS_PIN    PB14

MD_Parola P = MD_Parola(DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);

#define ARRAY_SIZE(x)  (sizeof(x)/sizeof(x[0]))

// Global data
typedef struct
{
  textEffect_t  effect;   // text effect to display
  char *        psz;      // text string nul terminated
  uint16_t      speed;    // speed multiplier of library default
  uint16_t      pause;    // pause multiplier for library default
} sCatalog;

sCatalog  catalog[] =
{
  { PA_PRINT, "PRINT", 1, 1 },
  { PA_FADE, "FADE", 20, 1 },
  { PA_WIPE, "WIPE", 5, 1 },
  { PA_WIPE_CURSOR, "WPE_C", 4, 1 },
  { PA_OPENING, "OPEN", 3, 1 },
  { PA_OPENING_CURSOR, "OPN_C", 4, 1 },
  { PA_CLOSING, "CLOSE", 3, 1 },
  { PA_CLOSING_CURSOR, "CLS_C", 4, 1 },
  { PA_BLINDS, "BLIND", 7, 1 },
  { PA_DISSOLVE, "DSLVE", 7, 1 },
  { PA_SCROLL_UP, "SC_U", 5, 1 },
  { PA_SCROLL_DOWN, "SC_D", 5, 1 },
  { PA_SCROLL_LEFT, "SC_L", 5, 1 },
  { PA_SCROLL_RIGHT, "SC_R", 5, 1 },
  { PA_SCROLL_UP_LEFT, "SC_UL", 7, 1 },
  { PA_SCROLL_UP_RIGHT, "SC_UR", 7, 1 },
  { PA_SCROLL_DOWN_LEFT, "SC_DL", 7, 1 },
  { PA_SCROLL_DOWN_RIGHT, "SC_DR", 7, 1 },
  { PA_SCAN_HORIZ, "SCNH", 4, 1 },
  { PA_SCAN_HORIZX, "SCNHX", 4, 1 },
  { PA_SCAN_VERT, "SCNV", 3, 1 },
  { PA_SCAN_VERTX, "SCNVX", 3, 1 },
  { PA_GROW_UP, "GRW_U", 7, 1 },
  { PA_GROW_DOWN, "GRW_D", 7, 1 },
};

void setup(void)
{
  delay(5000);
  P.begin(2);
  P.setInvert(false);

  P.setZone(0, 0, 15);
  P.setZone(1, 16, 31);
  P.setFont(0, BigFontBottom);
  P.setFont(1, BigFontUp);

  for (uint8_t i = 0; i < ARRAY_SIZE(catalog); i++)
  {
    catalog[i].speed *= P.getSpeed();
    catalog[i].pause *= 1000;
  }
   P.setIntensity(0);
}

void loop(void)
{
  for (uint8_t j = 0; j < 3; j++)
  {
    textPosition_t  just;

    switch (j)
    {
      case 0: just = PA_LEFT;    break;
      case 1: just = PA_CENTER;  break;
      case 2: just = PA_RIGHT;   break;
    }

    for (uint8_t i = 0; i < ARRAY_SIZE(catalog); i++)
    {
      P.displayClear();
      P.setFont(0, BigFontBottom);
      P.setFont(1, BigFontUp);
      P.displayZoneText(0, "WELCOME", just, catalog[i].speed, catalog[i].pause, catalog[i].effect, catalog[i].effect);
      P.displayZoneText(1, "WELCOME", just, catalog[i].speed, catalog[i].pause, catalog[i].effect, catalog[i].effect);
      while (!P.getZoneStatus(0) || !P.getZoneStatus(1))
        P.displayAnimate(); // animates and returns true when an animation is completed
      P.synchZoneStart();
      delay(catalog[i].pause);
    }
  }
}
when i'm scrolling the data to left or right,the bottom zone is scrolling slight faster than upper zone...,the two zone's are not sychronized properly.....

thanq 
Sankar

MajicDesigns, 2018-04-11:
What about the other question I asked:

> What is your matrix arrangement?

Also, does the text line up if you just print the top and bottom?

sankaramp, 2018-04-12:
Sorry i forgot,,my Matrix arragement

/**
 \def USE_FC16_HW
 Set to 1 to use FC16 hardware module kits.
 FC16 modules are similar in format to the ICStation modules but are wired differently.
 Modules are identified by a FC-16 designation on the PCB
  */
#define	USE_FC16_HW	1

sankaramp, 2018-04-12:
7654
3210

sankaramp, 2018-04-12:
total 32  8x8 matrices..
The Arrangement Stacked like this,

31   30   29   28   27   26   25   24   23   22   21   20   19   18   17   16
15   14   13   12   11   10   9     8     7     6     5     4     3     2     1     0

bottom 16 matrices consider as Bottom Zone &
Upper 16 matrices consider as Upper Zone

MajicDesigns, 2018-04-13:
The test code below run on my system keeps the two halves synchronized. What is clear when you run the animation really slowly, as per delays in this code, is that zone 0 is refreshed before zone 1, but at the end of each cycle they are aligned correctly. This effect will become more pronounced as more modules are added in each zone, as the comms/updates length are proportional to the number of modules.
`
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
#include "Font_Data.h"

#define MAX_DEVICES 22
#define CLK_PIN  13 // was PB6
#define DATA_PIN 11 // was PB7
#define CS_PIN   10 // was PB14

MD_Parola P = MD_Parola(DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);

#define ARRAY_SIZE(x) (sizeof(x)/sizeof(x[0]))

// Global data
typedef struct
{
  textEffect_t effect; // text effect to display
  char * psz; // text string nul terminated
  uint16_t speed; // speed multiplier of library default
  uint16_t pause; // pause multiplier for library default
} sCatalog;

sCatalog catalog[] =
{
/*  { PA_PRINT, "PRINT", 1, 1 },
  { PA_FADE, "FADE", 20, 1 },
  { PA_WIPE, "WIPE", 5, 1 },
  { PA_WIPE_CURSOR, "WPE_C", 4, 1 },
  { PA_OPENING, "OPEN", 3, 1 },
  { PA_OPENING_CURSOR, "OPN_C", 4, 1 },
  { PA_CLOSING, "CLOSE", 3, 1 },
  { PA_CLOSING_CURSOR, "CLS_C", 4, 1 },
  { PA_BLINDS, "BLIND", 7, 1 },
  { PA_DISSOLVE, "DSLVE", 7, 1 },
  { PA_SCROLL_UP, "SC_U", 5, 1 },
  { PA_SCROLL_DOWN, "SC_D", 5, 1 },
*/
  { PA_SCROLL_LEFT, "SC_L", 50, 5 },
  { PA_SCROLL_RIGHT, "SC_R", 50, 5 },
/*  { PA_SCROLL_UP_LEFT, "SC_UL", 7, 1 },
  { PA_SCROLL_UP_RIGHT, "SC_UR", 7, 1 },
  { PA_SCROLL_DOWN_LEFT, "SC_DL", 7, 1 },
  { PA_SCROLL_DOWN_RIGHT, "SC_DR", 7, 1 },
  { PA_SCAN_HORIZ, "SCNH", 4, 1 },
  { PA_SCAN_HORIZX, "SCNHX", 4, 1 },
  { PA_SCAN_VERT, "SCNV", 3, 1 },
  { PA_SCAN_VERTX, "SCNVX", 3, 1 },
  { PA_GROW_UP, "GRW_U", 7, 1 },
  { PA_GROW_DOWN, "GRW_D", 7, 1 },
*/};

void setup(void)
{
  P.begin(2);
  P.setZone(0, 0, (MAX_DEVICES/2) - 1);
  P.setZone(1, MAX_DEVICES/2, MAX_DEVICES-1);

  P.setFont(0, BigFontLower);
  P.setFont(1, BigFontUpper);

  for (uint8_t i = 0; i < ARRAY_SIZE(catalog); i++)
  {
    catalog[i].speed *= P.getSpeed();
    catalog[i].pause *= 1000;
  }
}

void loop(void)
{
  for (uint8_t j = 0; j < 3; j++)
  {
    textPosition_t just;

    switch (j)
    {
      case 0: just = PA_LEFT;    break;
      case 1: just = PA_CENTER;  break;
      case 2: just = PA_RIGHT;   break;
    }

    for (uint8_t i = 0; i < ARRAY_SIZE(catalog); i++)
    {
      P.displayClear();
      P.displayZoneText(0, "Welcome", just, catalog[i].speed, catalog[i].pause, catalog[i].effect, catalog[i].effect);
      P.displayZoneText(1, "Welcome", just, catalog[i].speed, catalog[i].pause, catalog[i].effect, catalog[i].effect);
      P.synchZoneStart();
      while (!P.getZoneStatus(0) || !P.getZoneStatus(1))
        P.displayAnimate(); // animates and returns true when an animation is completed
      delay(catalog[i].pause);
    }
  }
}
`

closed 2018-04-13 by MajicDesigns

sankaramp, 2018-04-13:
thanq for reply,

When speed is low, its working fyn,.,but when i increase the speed the same problem is arised.

my requirement is scrolling with high speed.
thanq.