MajicDesigns/MD_Parola #24: Errors compiling for ESP8266

ComputerZOO, 2018-05-05:
Good morning,
I got errors while compiling the sprite-demo for an ESP8266:
```
const uint8_t F_ROCKET = 2;
const uint8_t W_ROCKET = 11;
static const uint8_t rocket[F_ROCKET * W_ROCKET] PROGMEM = {
  0x18, 0x24, 0x42, 0x81, 0x99, 0x18, 0x99, 0x18, 0xa5, 0x5a, 0x81,
  0x18, 0x24, 0x42, 0x81, 0x18, 0x99, 0x18, 0x99, 0x24, 0x42, 0x99,
};

...

parola.setSpriteData(rocket, W_ROCKET, F_ROCKET, rocket, W_ROCKET, F_ROCKET);

error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
 parola.setSpriteData(rocket, W_ROCKET, F_ROCKET, rocket, W_ROCKET, F_ROCKET);
```
Does anybody has a solution for that? Everything else is running perfect.

MajicDesigns, 2018-05-05:
This seems he same as issue #23 already fixed. Please download the latest version.

closed 2018-05-06 by MajicDesigns

mdellen, 2018-07-28:
Same problem still in version=3.0.1 (platformio)

src/main.cpp:207:75: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
P.setSpriteData(rocket, W_ROCKET, F_ROCKET, rocket, W_ROCKET, F_ROCKET);
^

MajicDesigns, 2018-08-03:
Fixed