MajicDesigns/MD_Parola #25: Parola Zone problem reported on Arduino forum

MajicDesigns, 2018-05-06:
See details at [http://forum.arduino.cc/index.php?topic=545534.0](url)

walterdkn, 2018-05-11:
Hi,
I just want to let you know that I am playig around with the Parola_Timemsg example, I changed it to 4 zones like this:
 
  #define MAX_DEVICES 32

 ...
  P.setZone(0, 0, 7);
  P.setZone(1, 8, 15);
  P.setFont(1, numeric7Seg);
  P.setZone(2, 16, 23);
  P.setZone(3, 24, 31);

  P.displayZoneText(3, szTime, PA_CENTER, SPEED_TIME, PAUSE_TIME, PA_PRINT, PA_NO_EFFECT);
  P.displayZoneText(2, szMesg, PA_CENTER, SPEED_TIME, 0, PA_SCROLL_RIGHT, PA_SCROLL_RIGHT);
  P.displayZoneText(1, szTime, PA_CENTER, SPEED_TIME, PAUSE_TIME, PA_PRINT, PA_NO_EFFECT);
  P.displayZoneText(0, szMesg, PA_CENTER, SPEED_TIME, 0, PA_SCROLL_LEFT, PA_SCROLL_LEFT);
....

Interesting thing is this works (however, zone 2 stays blank, why ever...); at least there are no crashes and time display is updated in both zones 1 and 3.
[Parola_Zone_TimeMsg_mod.zip](https://github.com/MajicDesigns/MD_Parola/files/1996077/Parola_Zone_TimeMsg_mod.zip)

Maybe this helps...

Greetings,
Walter

Dancopy, 2018-11-09:
> 
> 
> Hi,
> I just want to let you know that I am playig around with the Parola_Zone_TimeMsg example, I changed it to 4 zones like this:
> 
> #define MAX_DEVICES 32
> 
> ...
> P.setZone(0, 0, 7);
> P.setZone(1, 8, 15);
> P.setFont(1, numeric7Seg);
> P.setZone(2, 16, 23);
> P.setZone(3, 24, 31);
> 
> P.displayZoneText(3, szTime, PA_CENTER, SPEED_TIME, PAUSE_TIME, PA_PRINT, PA_NO_EFFECT);
> P.displayZoneText(2, szMesg, PA_CENTER, SPEED_TIME, 0, PA_SCROLL_RIGHT, PA_SCROLL_RIGHT);
> P.displayZoneText(1, szTime, PA_CENTER, SPEED_TIME, PAUSE_TIME, PA_PRINT, PA_NO_EFFECT);
> P.displayZoneText(0, szMesg, PA_CENTER, SPEED_TIME, 0, PA_SCROLL_LEFT, PA_SCROLL_LEFT);
> ....
> 
> Interesting thing is this works (however, zone 2 stays blank, why ever...); at least there are no crashes and time display is updated in both zones 1 and 3.
> [Parola_Zone_TimeMsg_mod.zip](https://github.com/MajicDesigns/MD_Parola/files/1996077/Parola_Zone_TimeMsg_mod.zip)
> 
> Maybe this helps...
> 
> Greetings,
> Walter

Hello walterdkn!
Were you able to solve your 'zone 2 blank' problem? I would like, with your permission, to use your code!

MajicDesigns, 2019-01-30:
No similar issues reported.

closed 2019-01-30 by MajicDesigns

malekhosseini, 2020-10-15:
Hi
I have same problem with four zone. Buffer of data only have 2 zone data.My code is very complicate. I test above code and get same result.

walterdkn, 2020-10-16:
Hi malekhosseini,
in the meantime I reworked the code. This one works for me, a 4-line clock display (time, date, air pressure, temperature). You must take care of variable length in the arrays (e.g. monat[12][12]. I use an Arduino Mega for this, because with a Nano it didn't work good.
Greets,
walterdkn

[GPS_LED_PAROLA_MEGA_201001.zip](https://github.com/MajicDesigns/MD_Parola/files/5390384/GPS_LED_PAROLA_MEGA_201001.zip)

malekhosseini, 2020-10-16:
Hi walterdkn
Thank you.
This is not my issues because i use ESP32 with 320k RAM.

Dancopy, 2020-10-16:
Hello walterdkn
1) Could you post a video on youtube about what your code does and what components are you using?
2) I don't understand why you are using FastLED? Aren't you using the FC16_HW Matrix?
Thank you

Dancopy, 2020-10-20:
Waiting for an answer!

walterdkn, 2020-10-21:
Hi Dancopy,
my reply via email seems not to have reached you. So here is what I wrote and a link to a YT video at the end:

The clock has 60 neopixels for the display of seconds around the matrices. I use the following components:

- Arduino Mega
- 16x MAX7219 dot matrix displays, arranged in 4 lines with 4 displays each
- 60x WS2812 (12 stripes with 5 pixels each; however, single LEDs look much better since you can arrange them in a perfect circle; I made this in other projects)
- NEO-6M GPS module for receiving time and date; in the sketch the UTC time is converted to local time with daylight saving dates
- BME280 sensor for air pressure and temperature
- BH1750 light sensor for automatic dimming; however, dimming of the matrices is way limited, even with the lowest setting the display are very bright.
- 5V/2500mA wall plug; if you do not make use of full brightness of matrices and/or neopixels this is sufficient.

1st line: static display of time (hh:mm)
2nd line: running display of date (weekday, day, written month, year)
3rd line: static display of air pressure (in hPa)
4th line: static display of temperature (degrees celsius)

I hope this helps a bit. The arduino code is not yet optimized, but runs so far without errors.

Here is a link to a short YT video: https://www.youtube.com/watch?v=qBOvnark3so

Greetings
walterdkn

Dancopy, 2020-10-21:
Wow! Very cool your project, congratulations!