MajicDesigns/MD_Parola #30: 'class MD_Parola' has no member named 'getIntensity'

ArduChecker, 2018-08-08:
Hi,
   the following statement within arduino code 
     `uint8_t  scrollIntensity = P.getIntensity();`
  ends up in the error message see title.

 Strange, usage of 
`uint16_t  scrollSpeed = P.getSpeed();`
causes no problems, the usage of P.getCharSpacing() is working, too.

Checked up MD.Parola.h and found in class definition...
`
  /**
  * Get the zone brightness.
  *
  * Get the intensity (brightness) of the display.
  *
  * \return The intensity setting.
  */
  inline uint8_t getIntensity() { return _intensity; }`

right below method getCharSpacing....

Used #include and #define 

#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
.....

#define HARDWARE_TYPE MD_MAX72XX::FC16_HW 
#define MAX_DEVICES 4               // Number of 8x8 modules linked together
#define CLK_PIN   13                // or SCK
#define DATA_PIN  14                // or MOSI
#define CS_PIN    12                // or SS

// SOFTWARE SPI
MD_Parola P = MD_Parola(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);

MajicDesigns, 2018-08-14:
Parola library compiles all examples ok and no response fro OP, closing issue.

closed 2018-08-14 by MajicDesigns

Martin12350, 2023-06-22:
I am guessing not a lot of people is trying to read intensity back, but same error here ;)
`int intensity = Display.getIntensity();`
Compilation error: 'class MD_Parola' has no member named 'getIntensity'.

In source code You have only:
`inline uint8_t getIntensity() { return _intensity; }`

And nothing else, while e.g. getPause has void as parameter and then 2 another functions.
Same for speed etc.

Can You please check if this really works? Thanks a lot.

Martin12350, 2023-06-22:
I was just comparing how other functions in the library work. Like pause, You can have each zone with a different pause, but if You call getPause without a parameter, You get zone 0.
Also in my case, when I have only a few modules, I do not use zones, so I call all functions without specifying zone, which automatically results in 0.
Anyway, how You deal with it is up to You. But would be cool to have getIntensity in MD_P.
Keep the good work :)

MajicDesigns, 2023-07-03:
versin 3.7.1 makes all this visible and symmetric with other functions.