MajicDesigns/MD_Parola #31: Libarary too big for Adafruit Feather 32u4

tavdog, 2018-08-13:
I need to reduce the size of either MD_Parola or the MAX_72XX libarary by 1372 bytes.  Is there a previous version that is smaller? Is there another way to make it just a little bit smaller ?

thanks.

MajicDesigns, 2018-08-13:
Couple of things that may help:
1. There are a number of #defines in the Parola header file that allow you to turn off animation groups that you may not be using.
`// Granular selection of animations/functions to include in the library
// If an animation class is not used at all some memory savings can be made
// by excluding the animation code.
#define ENA_MISC    1   ///< Enable miscellaneous animations
#define ENA_WIPE    1   ///< Enable wipe type animations
#define ENA_SCAN    1   ///< Enable scanning animations
#define ENA_SCR_DIA 1   ///< Enable diagonal scrolling animation
#define ENA_OPNCLS  1   ///< Enable open and close scan effects
#define ENA_GROW    1   ///< Enable grow effects
#define ENA_SPRITE  1   ///< Enable sprite effects
`
2. In the MD_MAX72xx font file, the new version 3+ font definitions allow for smaller font files if you are only using a subset of the ASCII characters. You can set a start ASCII code and and end ASCII code that are in the font file. See the MD_MAX72xx doc or the header file for a description of the font file. You will need to edit the first line of the font data and change the number of table entries to match the definition. getting rid of the characters with code > 127 could get quite a bit of flash RAM back.

tavdog, 2018-08-13:
Thank you so much. I'll try it.

closed 2018-08-13 by tavdog