MajicDesigns/MD_Parola #32: It says 'Setup' Cannot find a way to setup.

naidu92, 2018-11-23:
Respected Sir/Madam,
I've uploaded this code to Arduino Mega, no problem in uploading the code and also changed the pin number, bluetooth module number correctly. While it starts up, the Dot Matrix MAX7219 displays only 'Setup' I cannot find a way to setup the bluetooth. I've read the code and it said to write 'AT+' and something like 'NAME\0\0' or '\r\n' I cannot understand how to setup this bluetooth. Please help me do this setup thing. And I am using HC-06 module.
Thank you.

MajicDesigns, 2018-11-24:
I have no idea what code you are talking about, whose code you are using, and what you are expecting to happen.

If you need help to set up Bluetooth, then please use the Arduino forum at arduino.cc. If you want to report a bug in the Parola Library, then this is the place to do that.

naidu92, 2018-11-24:
This is the code : Parola_Bluetooth_Control. The example code from
MD_Parola library. Help me to setup bluetooth HC-06.

naidu92, 2018-11-25:
>    1. Make sure that the BT module works without the PArola stuff on top.
>    Lots of stuff on the internet on how to do this. This makes sure the
>    connections are correct.
>    2. Set the #define for HC06 to 1 and the other to 0, define the pins
>    for the SoftwareSerial library to be how you have connected the module. If
>    you did part 1 correctly, the Parola example should work.

I guessed the same issue that the module would be malfunctioning. But
thanks for suggestion. I will reply you if my thing works and send you a
video of that scrolling text display with bluetooth. Its been 9 months or
more than that I am working on that project without knowing how to code a
certain project. Mostly completed but the bluetooth part is remaining.
Thank you. It helps and means a lot.