MajicDesigns/MD_Parola #33: Can't define MD_Parola by Pointer

human890209, 2018-11-27:
Hi,
I want to make the instance defined by dynamic settings, so I try to use a pointer. But nothing will be shown on the screen.
I make two simple example to demonstrate the problem.
Example 1 Normal Way - Working
```
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>

#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 4

#define CLK_PIN   27
#define DATA_PIN  25
#define CS_PIN    32

MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);

void setup(void)
{
	SPI.begin(CLK_PIN, 0, DATA_PIN, 0);
  P.begin();
  P.displayText("Hello", PA_CENTER, 100, 500, PA_SCROLL_LEFT, PA_SCROLL_LEFT);
}

void loop(void)
{
  P.displayAnimate();
}
```

Example 2 Pointer Way - Display Nothing
```
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>

#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 4

#define CLK_PIN   27
#define DATA_PIN  25
#define CS_PIN    32

MD_Parola* P;

void setup(void)
{
	P = new MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);

	SPI.begin(CLK_PIN, 0, DATA_PIN, 0);
	P->begin();
	P->displayText("Hello", PA_CENTER, 100, 500, PA_SCROLL_LEFT, PA_SCROLL_LEFT);
}

void loop(void)
{
	P->displayAnimate();
}
```

I'm not a good coder. 
Hope someone could tell me why Pointer won't work and How to modify it to make pointer work?

human890209, 2018-11-27:
I initialize all the variables in the Constructor, and the problem is fixed. Example 2 working!

closed 2018-11-27 by human890209

MajicDesigns, 2018-11-27:
Unrelated, but in your code you do not need to initialise the SPI library. That is done in the MD_MAX72xx library already.