MajicDesigns/MD_Parola #35: Indent from a character of zero length / Отступ от символа нулевой длинны

immortalserg, 2018-12-23:
When using characters of zero length, a 1 pixel gap is inserted after the character with zero length. I do this to use the Russian font without transcoding, for this I created a font where the most significant bit is zero length (208 and 209), works well, but it turns out that there is a gap of 2 pixels between the characters. It does not have any logic. Is it possible to fix this?
Sketch and font here: https://github.com/immortalserg/MD_Parola_fonts

При использовании символов нулевой длинны после символа с нулевой длинной вставляется разрыв в 1 пиксель. Я это делаю для использования русского шрифта без перекодировки, для этого создал шрифт где старший бит нулевой длинны (208 и 209), работает хорошо но получается, что между символами разрыв в 2 пикселя. Это не имеет ни какой логики. Возможно ли это исправить?
Скетч и шрифт тут: https://github.com/immortalserg/MD_Parola_fonts

MajicDesigns, 2018-12-23:
Fixed. Release as version 3.0.2

immortalserg, 2018-12-23:
after the update, it does not display characters at all that are used without conversion in UTP-8.
The letter "Б" is UTF-8 encoding 208, 145
in the library version 3.0.1 string
`P.displayText ("ББ", PA_CENTER, 0, 0, PA_PRINT, PA_NO_EFFECT);`
displayed characters: 208 145 208 145
I use the symbol 145 in the character code 145
`4, 127, 73, 73, 49, // 145 - Б`
in character code 208 - empty
`0, // 208 - for rus fonts UTF-8`
displays between 2 letters Б two intervals.
after updating to 3.0.2 does not display characters with code 145
This sketch does not display data in version 3.0.2, in version 3.0.1 data displayshttps://github.com/immortalserg/MD_Parola_fonts/tree/master/examples/Parola_Fonts

после обновления не отображает вообще символы которые без перекодировки в UTF-8 используются.
Буква "Б" имеет UTF-8 кодировку 208, 145
в версии библиотеки 3.0.1 строка 
`P.displayText("ББ", PA_CENTER, 0, 0, PA_PRINT, PA_NO_EFFECT);`
выводила символы: 208 145 208 145
я в шрифте использую в коде символа 145 символ "Б"
`4, 127, 73, 73, 49,	// 145 - Б`
в коде символа 208 - пустой
`0,	// 208 - for rus fonts UTF-8`
выводит между 2 буквами Б два интервала.
после обновления на 3.0.2 не выводит символы с кодом 145 в шрифте.
Этот скетч не выводит данные в версии 3.0.2, в версии 3.0.1 данные выводит https://github.com/immortalserg/MD_Parola_fonts/tree/master/examples/Parola_Fonts

MajicDesigns, 2018-12-24:
Should be fixed now. Release 3.1.0
Zero length character was taken to be the end of the string, which was your first character. Animations PRINT, HSCROLL and SLICE needed logic changes modification to allow the zero width length characters + some internal changes to the lib. More work that I thought was necessary. :-(

immortalserg, 2018-12-24:
running, the problem is closed

closed 2018-12-24 by immortalserg