MajicDesigns/MD_Parola #36: Detect end of string

michelep, 2018-12-25:
How i can detect the end of a string scrolling ? I.E. to detect when i can change the display string ?

michelep, 2018-12-26:
Sorry, i'm using MD_MAX72XX library and displayAnimate() seems to be not included.

MajicDesigns, 2018-12-26:
You raised the issue in the MD_Parola repository, so you got an answer for MD_Parola.

With MD_MAX72xx there is nothing to help you, you need to work it out for yourself by keeping track of where the text is. This should not be hard as you are doing all the scrolling animation in your code anyway.

In the MD_Parola code, I count all the empty columns in the display 'after' (depends on which direction is being scrolled) the text being displayed . When these empty columns equal number of columns in the display the text has completely scrolled off and the animation is complete.