MajicDesigns/MD_Parola #37: how to use zones and normal at the same time?

wchpikus, 2018-12-28:
hi
esp 8622, md_parola 3.0.2
i have 12 display line and 3 columns, 36 matrix display.
can i use 2 for big font (from exampe) and one normal?
i try to display some text in 2'th zone, but something wrong..

0 zone and 1 zone is fordouble height, 2 for normal.
#define MAX_DEVICES 36
#define MAX_ZONES 3
#define MAX_DEVICES (MAX_ZONES * ZONE_SIZE)
#define ZONE_SIZE 12
#define ZONE_UPPER  0
#define ZONE_LOWER  1
#define ZONE_FREE 2

  P.setZone(ZONE_UPPER,0, ZONE_SIZE - 1 );
  P.setZone(ZONE_LOWER,ZONE_SIZE,(ZONE_SIZE*2)-1 );
  P.setZone(ZONE_FREE, (ZONE_SIZE*2), MAX_DEVICES);

in loop ::
    P.displayClear();
      P.displayZoneText(ZONE_UPPER, msgH, PA_CENTER, SCROLL_SPEED, PAUSE_TIME, scrollUpper, scrollLower);
      P.displayZoneText(ZONE_LOWER, msgL[cycle], PA_CENTER, SCROLL_SPEED, PAUSE_TIME, scrollUpper, scrollLower);
      P.displayZoneText(ZONE_FREE,"test", PA_CENTER, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_SCROLL_LEFT);
      cycle = (cycle + 1) % ARRAY_SIZE(msgL);
      P.synchZoneStart();
  
  }

and "test" take big font..can i use normal font for this?
I do not see any example for use gig font and small at the same time.

add new fonts?
P.displayClear();
      P.setFont(BigFont);
      P.displayZoneText(ZONE_UPPER, msgH, PA_CENTER, SCROLL_SPEED, PAUSE_TIME, scrollUpper, scrollLower);
      P.displayZoneText(ZONE_LOWER, msgL[cycle], PA_CENTER, SCROLL_SPEED, PAUSE_TIME, scrollUpper, scrollLower);
      P.setFont(font);
      P.displayZoneText(ZONE_FREE,"test", PA_CENTER, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_SCROLL_LEFT);
      cycle = (cycle + 1) % ARRAY_SIZE(msgL);
      P.synchZoneStart();

MajicDesigns, 2018-12-28:
> i try to display some text in 2'th zone, but something wrong..
What exactly is wrong? What do you see and what are you expecting to see?

Please use the <> to post your code as it is not being displayed properly - edit your original post. And it is better to post all the code instead of code fragments. Often the error is not where you think it is..

wchpikus, 2018-12-28:
hi
i like to on 1 and 2 line have big fonts, on 3 line 8x5 `fonts
`
#include <ESP8266WiFi.h>
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
#include "Font_Data.h"
#include "fonts.h"

// Turn debugging on and off
#define DEBUG 1

#if DEBUG
#define PRINTS(s)   { Serial.print(F(s)); }
#define PRINT(s, v) { Serial.print(F(s)); Serial.print(v); }
#else
#define PRINTS(s)
#define PRINT(s, v)
#endif

// Define the main direction for scrolling double height.
// if 1, scroll left; if 0, scroll right
#define SCROLL_LEFT 1

// Hardware adaptation parameters for scrolling
bool invertUpperZone = false;
textEffect_t scrollUpper, scrollLower;

// Define the number of devices we have in the chain and the hardware interface
// NOTE: These pin numbers may not work with your hardware and may need changing
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 36
#define MAX_ZONES 3
#define MAX_DEVICES (MAX_ZONES * ZONE_SIZE)
#define ZONE_SIZE 12
#define ZONE_UPPER  0
#define ZONE_LOWER  1
#define ZONE_FREE 2
#define PAUSE_TIME 0
#define SCROLL_SPEED 30

#define  CLK_PIN   14
#define DATA_PIN  13
#define CS_PIN    15

// HARDWARE SPI
MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);

#define ARRAY_SIZE(a) (sizeof(a)/sizeof(a[0]))
char *msgL[] =
{
  "Double height with custom font & 2 zones",
  "ABCDEFGHIJKLMNOPQRSTUVWXYZ",
  "abcdefghijklmnopqrstuvwxyz",
  "0123456789",
  "`!@#$%^&*()_+-={};:'<>\"?,./|\\{}",
};
char *msgH; // allocated memory in setup()

void setup(void)
{
  uint8_t max = 0;
  P.setIntensity(0);

#if DEBUG
  Serial.begin(115200);
  PRINTS("\n[Double_Height_v2]");
#endif

  // set up global parameters
  invertUpperZone = (HARDWARE_TYPE == MD_MAX72XX::GENERIC_HW || HARDWARE_TYPE == MD_MAX72XX::PAROLA_HW);
  if (invertUpperZone)
  {
#if SCROLL_LEFT // invert and scroll left
    scrollUpper = PA_SCROLL_RIGHT;
    scrollLower = PA_SCROLL_LEFT;
#else           // invert and scroll right
    scrollUpper = PA_SCROLL_LEFT;
    scrollLower = PA_SCROLL_RIGHT;
#endif
  }
  else // not invert
  {
#if SCROLL_LEFT // not invert and scroll left
    scrollUpper = PA_SCROLL_LEFT;
    scrollLower = PA_SCROLL_LEFT;
#else           // not invert and scroll right
    scrollUpper = PA_SCROLL_RIGHT;
    scrollLower = PA_SCROLL_RIGHT;
#endif
  }

  // work out the size of buffer required
  for (uint8_t i = 0; i<ARRAY_SIZE(msgL); i++)
    if (strlen(msgL[i]) > max) max = strlen(msgL[i]);

  msgH = (char *)malloc(sizeof(char)*(max + 2));

  // initialise the LED display
  P.begin(MAX_ZONES);

  // Set up zones for 2 halves of the display
  P.setZone(ZONE_UPPER,0, ZONE_SIZE - 1 );
  P.setZone(ZONE_LOWER,ZONE_SIZE,(ZONE_SIZE*2)-1 );
  P.setZone(ZONE_FREE, (ZONE_SIZE*2), MAX_DEVICES);

//  P.setZone(ZONE_UPPER,0,11 );
//  P.setZone(ZONE_LOWER,12,24 );
//  P.setZone(ZONE_FREE,25,36);
  
//  P.setFont(BigFont);
  P.setCharSpacing(P.getCharSpacing() * 2); // double height --> double spacing
  if (invertUpperZone)
  {
    P.setZoneEffect(ZONE_UPPER, true, PA_FLIP_UD);
    P.setZoneEffect(ZONE_UPPER, true, PA_FLIP_LR);
  }
  PRINT("\nFLIP_UD=", P.getZoneEffect(ZONE_UPPER, PA_FLIP_UD));
  PRINT("\nFLIP_LR=", P.getZoneEffect(ZONE_UPPER, PA_FLIP_LR));
  PRINT("\nSCROLL_LEFT=", SCROLL_LEFT);
}

void createHString(char *pH, char *pL)
{
  for (; *pL != '\0'; pL++)
    *pH++ = *pL | 0x80;   // offset character

  *pH = '\0'; // terminate the string
}

void loop(void)
{
  static uint8_t cycle = 0;

  P.displayAnimate();

  if (P.getZoneStatus(ZONE_LOWER) && P.getZoneStatus(ZONE_UPPER) && P.getZoneStatus(ZONE_FREE))
  {
    PRINT("\n", cycle);
    PRINT(": ", msgL[cycle]);

    // set up the string
    createHString(msgH, msgL[cycle]);

    P.displayClear();
      P.setFont(BigFont);
      P.displayZoneText(ZONE_UPPER, msgH, PA_CENTER, SCROLL_SPEED, PAUSE_TIME, scrollUpper, scrollLower);
      P.displayZoneText(ZONE_LOWER, msgL[cycle], PA_CENTER, SCROLL_SPEED, PAUSE_TIME, scrollUpper, scrollLower);
      P.synchZoneStart();
      test();

      cycle = (cycle + 1) % ARRAY_SIZE(msgL);
//      P.synchZoneStart();
    
  }
}

void test(){
      P.setFont(font);
      P.displayZoneText(ZONE_FREE,"......test........", PA_CENTER, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_SCROLL_LEFT);  
}`

MajicDesigns, 2018-12-28:
Please see my previous comment. 
You need to tell me what is not working. I do not have 36 modules nor do I have the time to set this up and try your software. If you tell me what you see and what is not working I may be able to tell from the code. Otherwise this will never be resolved.

` (P.getZoneStatus(ZONE_LOWER) && P.getZoneStatus(ZONE_UPPER) && P.getZoneStatus(ZONE_FREE)`
This will NOT work as the LOWER and UPPER follow the same time for animation while FREE must be on different timing as it is a different size. You need to check for FREE separately.

wchpikus, 2018-12-28:
Ok i see.
And how to change font? For 16x8 you used BigFonts.

MajicDesigns, 2018-12-28:
Every zone has its own font. You set it the same way you set the font in the other zones. If you don't set the font (or set it to nullptr) you get the standard font.
When you don't specify the number of zones you are actually running a one zone display.

wchpikus, 2018-12-29:
ok
code:
`// Demonstrates one double height display using Parola using a single font file
// defintion created with the MD_MAX72xx font builder.
//
// Each font file has the lower part of a character as ASCII codes 0-127 and the
// upper part of the character in ASCII code 128-255. Adding 128 to each lower
// character creates the correct index for the upper character.
// The upper and lower portions are managed as 2 zones 'stacked' on top of each other
// so that the module numbers are in the sequence shown below:
//
// * Modules (like FC-16) that can fit over each other with no gap
//  n n-1 n-2 ... n/2+1   <- this direction top row
//  n/2 ... 3  2  1  0    <- this direction bottom row
//
// * Modules (like Generic and Parola) that cannot fit over each other with no gap
//  n/2+1 ... n-2 n-1 n   -> this direction top row
//  n/2 ... 3  2  1  0    <- this direction bottom row
//
// Sending the original string to the lower zone and the modified (+128) string to the
// upper zone creates the complete message on the display.
//
// MD_MAX72XX library can be found at https://github.com/MajicDesigns/MD_MAX72XX
//
#include <ESP8266WiFi.h>
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
#include "Font_Data.h"
#include "fonts.h"

// Turn debugging on and off
#define DEBUG 1

#if DEBUG
#define PRINTS(s)   { Serial.print(F(s)); }
#define PRINT(s, v) { Serial.print(F(s)); Serial.print(v); }
#else
#define PRINTS(s)
#define PRINT(s, v)
#endif

// Define the main direction for scrolling double height.
// if 1, scroll left; if 0, scroll right
#define SCROLL_LEFT 1

// Hardware adaptation parameters for scrolling
bool invertUpperZone = false;
textEffect_t scrollUpper, scrollLower;

// Define the number of devices we have in the chain and the hardware interface
// NOTE: These pin numbers may not work with your hardware and may need changing
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 36
#define MAX_ZONES 3
#define MAX_DEVICES (MAX_ZONES * ZONE_SIZE)
#define ZONE_SIZE 12
#define ZONE_UPPER  0
#define ZONE_LOWER  1
#define ZONE_FREE 2
#define PAUSE_TIME 0
#define SCROLL_SPEED 30

#define  CLK_PIN   14
#define DATA_PIN  13
#define CS_PIN    15

// HARDWARE SPI
MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);

#define ARRAY_SIZE(a) (sizeof(a)/sizeof(a[0]))
char *msgL[] =
{
  "Double height",
  "TEST",
  "test",
  "01234",
  "`!@#$",
};
char *msgH; // allocated memory in setup()

void setup(void)
{
  uint8_t max = 0;
  P.setIntensity(0);

  // set up global parameters
  invertUpperZone = (HARDWARE_TYPE == MD_MAX72XX::GENERIC_HW || HARDWARE_TYPE == MD_MAX72XX::PAROLA_HW);
  if (invertUpperZone)
  {
#if SCROLL_LEFT // invert and scroll left
    scrollUpper = PA_SCROLL_RIGHT;
    scrollLower = PA_SCROLL_LEFT;
#else           // invert and scroll right
    scrollUpper = PA_SCROLL_LEFT;
    scrollLower = PA_SCROLL_RIGHT;
#endif
  }
  else // not invert
  {
#if SCROLL_LEFT // not invert and scroll left
    scrollUpper = PA_SCROLL_LEFT;
    scrollLower = PA_SCROLL_LEFT;
#else           // not invert and scroll right
    scrollUpper = PA_SCROLL_RIGHT;
    scrollLower = PA_SCROLL_RIGHT;
#endif
  }

  // work out the size of buffer required
  for (uint8_t i = 0; i<ARRAY_SIZE(msgL); i++)
    if (strlen(msgL[i]) > max) max = strlen(msgL[i]);

  msgH = (char *)malloc(sizeof(char)*(max + 2));

  // initialise the LED display
  P.begin(MAX_ZONES);

  // Set up zones for 2 halves of the display
  P.setZone(ZONE_UPPER,0, ZONE_SIZE - 1 );            //  P.setZone(ZONE_UPPER,0,11 );
  P.setZone(ZONE_LOWER,ZONE_SIZE,(ZONE_SIZE*2)-1 );   //  P.setZone(ZONE_LOWER,12,24 );
  P.setZone(ZONE_FREE, ZONE_SIZE*2, (ZONE_SIZE*3)-1); //  P.setZone(ZONE_FREE,24,35);

  
//  P.setFont(BigFont);
  P.setCharSpacing(P.getCharSpacing() * 2); // double height --> double spacing
  if (invertUpperZone)
  {
    P.setZoneEffect(ZONE_UPPER, true, PA_FLIP_UD);
    P.setZoneEffect(ZONE_UPPER, true, PA_FLIP_LR);
  }
}

void createHString(char *pH, char *pL)
{
  for (; *pL != '\0'; pL++)
    *pH++ = *pL | 0x80;   // offset character

  *pH = '\0'; // terminate the string
}

void loop(void){
 static uint8_t cycle = 0;

P.displayAnimate();

if (P.getZoneStatus(ZONE_LOWER) && P.getZoneStatus(ZONE_UPPER))
  {
//    PRINT("\n", cycle);
//    PRINT(": ", msgL[cycle]);
    createHString(msgH, msgL[cycle]);

//    P.displayClear();
    P.setFont(BigFont);
    P.displayZoneText(ZONE_UPPER, msgH, PA_CENTER, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_SCROLL_LEFT);
    P.displayZoneText(ZONE_LOWER, msgL[cycle], PA_CENTER, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_SCROLL_LEFT);
    P.synchZoneStart();
    cycle = (cycle + 1) % ARRAY_SIZE(msgL);
  }

    if (P.getZoneStatus(ZONE_FREE)){
//      P.displayClear();
//      P.setFont(font);
      P.displayZoneText(ZONE_FREE, "test", PA_LEFT, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_SCROLL_RIGHT);
    }

}
`

works, but in zone 2 fonts are the same as upper..
https://www.youtube.com/watch?v=_QlbG0nVwJc

MajicDesigns, 2018-12-29:
`P.setFont(BigFont);`
Sets the font for ALL zones.

Use setZoneFont() to set the font for one zone.

wchpikus, 2018-12-29:
'class MD_Parola' has no member named 'setZoneFont'

MajicDesigns, 2018-12-29:
My mistake. setZoneFont is an internal function. The external function is setFont but you specify the zone you want.

All this is in the library documentation. You may save time if you read that.

wchpikus, 2018-12-29:
ok, so in this case:
``void loop(void){
 static uint8_t cycle = 0;

P.displayAnimate();

if (P.getZoneStatus(ZONE_LOWER) && P.getZoneStatus(ZONE_UPPER))
  {
    createHString(msgH, msgL[cycle]);
//    P.displayClear();
    P.setFont(0,BigFont);
    P.setFont(1,BigFont);
    P.displayZoneText(ZONE_UPPER, msgH, PA_RIGHT, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_SCROLL_DOWN_RIGHT);
    P.displayZoneText(ZONE_LOWER, msgL[cycle], PA_RIGHT, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_SCROLL_UP_RIGHT);
    P.synchZoneStart();
    cycle = (cycle + 1) % ARRAY_SIZE(msgL);
  }

    if (P.getZoneStatus(ZONE_FREE)){
//      P.displayClear();
      P.setFont(2,font5x7);
      P.displayZoneText(ZONE_FREE, "test", PA_LEFT, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_SCROLL_RIGHT);
    }
}``
P.setFont(2,font5x7);, yes?

MajicDesigns, 2018-12-29:
I would set the font in the setup() otherwise you are setting it up many many many more times than required. You only need to tell the library once if you don't plan to change it.

The location of the zone 2 setFont will mean it is executed many hundreds of times a second - no need, especially as there is some overhead in to setting the font.

wchpikus, 2018-12-29:
Yes,offcourse,  i taki it yo setup, this only for test.
Sure to call once when nothing change on zone..

I saw some strange behaviour.
From time to time, one or more of displays (random) stop working good or stay black..
After reset works ok.
Strange, this is no hardware isue, but i was check the scope of this signals and zonk..
Clock in the all out pins module looks like saw, no like rectangle..from esp looks ok.
Data looks similar..
Transmision error? China fake max chips?

MajicDesigns, 2018-12-29:
The problem may just be the number of devices in the chain, bad connections, power supply, signal attenuation, etc, but you have not provided any information on this. All the signals can be provided in parallel (see picture at https://arduinoplusplus.wordpress.com/2018/07/13/fun-and-games-and-a-new-library/#jp-carousel-9424) to the modules except for DataOut to DataIn, which should never have a bad signal anyway. If you do them all serially there are issues with attenuation and dropouts unless you take precautions. Lots of that on the internet, just search for it.

As your software issues are resolved I will close this issue.

closed 2018-12-29 by MajicDesigns

wchpikus, 2018-12-30:
tnx:)

wchpikus, 2018-12-30:
could i ask for one more?
i try to write function with string, to display text without table:
`char *msgL[] =
{
  "Double height",
  "TEST LARGE",
  "test small",
  "01234 43210",
  "!@#$ %$#@!",
};`

`void bigFont(String message){
if (P.getZoneStatus(ZONE_LOWER) && P.getZoneStatus(ZONE_UPPER))
  {
    int str_len = message.length() + 1;       //length of string
    char char_array[str_len];                 //buffer of length of string
    message.toCharArray(char_array, str_len); //copy to buffer
    
    createHString(msgH, char_array);          //create big font...what i have to do?
    P.setFont(0,BigFont);
    P.setFont(1,BigFont);
    P.displayZoneText(ZONE_UPPER, msgH, PA_RIGHT, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_CLOSING);
    P.displayZoneText(ZONE_LOWER, char_array, PA_RIGHT, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_CLOSING);
    P.synchZoneStart();
`

i see only upper display..

MajicDesigns, 2018-12-30:
I don't use Strings so hard to give any advice (other than learn not to use them on microcontrollers).
I suspect that the definitions for the size of the strings may be wrong. Try printing out the text and make sure it is what you expect it to be.

wchpikus, 2018-12-30:
Did you expect on the end of data /0 not 0 yes?

wchpikus, 2018-12-31:
ok, this way works ok:

#define MAX_MESG 20
char Mesg[MAX_MESG+1] = "";

bigFont("123456789");

`void bigFont(String message){
    int str_len = message.length()+1;       //length of string
    Serial.println(str_len);
    char char_array[str_len];                 //buffer of length of string
    message.toCharArray(char_array, str_len); //copy to buffer   
    strcpy(Mesg, char_array);
    createHString(msgH, char_array);          //create big font 
    P.setFont(0,BigFont);
    P.setFont(1,BigFont);
    P.displayZoneText(ZONE_UPPER, msgH, PA_CENTER, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_CLOSING);
    P.displayZoneText(ZONE_LOWER, Mesg, PA_RIGHT, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_CLOSING);
    P.synchZoneStart();
}`

wchpikus, 2019-01-02:
one more..
Why when text is smaller than zone working ok, but when is biggest, working strange?
https://www.youtube.com/watch?v=gJ3-wIHECIA
https://www.youtube.com/watch?v=E91-GtjxiKA

code:
`#include <ESP8266WiFi.h>
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
#include "Font_Data.h"
#include "fonts.h"

// Define the main direction for scrolling double height.
// if 1, scroll left; if 0, scroll right
#define SCROLL_LEFT 1

// Hardware adaptation parameters for scrolling
bool invertUpperZone = false;
textEffect_t scrollUpper, scrollLower;

// Define the number of devices we have in the chain and the hardware interface
// NOTE: These pin numbers may not work with your hardware and may need changing
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 36
#define MAX_ZONES 3
#define MAX_DEVICES (MAX_ZONES * ZONE_SIZE)
#define ZONE_SIZE 12
#define ZONE_UPPER  0
#define ZONE_LOWER  1
#define ZONE_FREE 2
#define PAUSE_TIME 1000 //in milliseconds
#define SCROLL_SPEED 20
#define MAX_MESG 20

#define  CLK_PIN   14
#define DATA_PIN  13
#define CS_PIN    15

// HARDWARE SPI
MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);

char MesgLARGE[MAX_MESG+1] = "";
char MesgSMALL[MAX_MESG+1] = "";
char *msgH; 
uint8_t str_len; //buffer size

void setup(void)
{
  Serial.begin(115200);
  delay(10);
  uint8_t max = 0;
  P.setIntensity(0);
  msgH = (char *)malloc(sizeof(char)*(max + 2));

  // initialise the LED display
  P.begin(MAX_ZONES);

  // Set up zones for 2 halves of the display
  P.setZone(ZONE_UPPER,0, ZONE_SIZE - 1 );            //  P.setZone(ZONE_UPPER,0,11 );
  P.setZone(ZONE_LOWER,ZONE_SIZE,(ZONE_SIZE*2)-1 );   //  P.setZone(ZONE_LOWER,12,24 );
  P.setZone(ZONE_FREE, ZONE_SIZE*2, (ZONE_SIZE*3)-1); //  P.setZone(ZONE_FREE,24,35);
//  P.displayZoneText(ZONE_UPPER, msgH, PA_CENTER, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_RIGHT, PA_SCROLL_LEFT);
//  P.displayZoneText(ZONE_LOWER, MesgLARGE, PA_CENTER, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_RIGHT, PA_SCROLL_LEFT);
//  P.displayZoneText(ZONE_FREE, MesgSMALL, PA_CENTER, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_SCROLL_LEFT);
  P.setCharSpacing(P.getCharSpacing() * 2); // double height --> double spacing
  
  if (invertUpperZone)
  {
    P.setZoneEffect(ZONE_UPPER, true, PA_FLIP_UD);
    P.setZoneEffect(ZONE_UPPER, true, PA_FLIP_LR);
  }
  P.displayClear();
}

void createHString(char *pH, char *pL)
{
  for (; *pL != '\0'; pL++)
    *pH++ = *pL | 0x80;   // offset character

  *pH = '\0'; // terminate the string
}

uint8_t i,y;
void loop(void){
P.displayAnimate();
    if (P.getZoneStatus(ZONE_FREE)){
      switch(i){
        case 0:
        smallFont("jeden test ");
        break;

        case 1:
        smallFont("dwa test ");   
        break;
        
        case 2:
        smallFont("trzy test ");
        break;
      }
      i++;
      P.displayReset(ZONE_FREE);
      if  (i>2){
        i=0;
      }      
}            
   
    
    
    if (P.getZoneStatus(ZONE_LOWER) && P.getZoneStatus(ZONE_UPPER)){
      P.setFont(ZONE_UPPER,BigFont);
      P.setFont(ZONE_LOWER,BigFont);    
      switch(y){
        case 0:
        bigFont("12345678910");
        P.synchZoneStart(); 
        break;

        case 1:
        bigFont("abcdefghijk");
        P.synchZoneStart();       
        break;
        
        case 2:
        bigFont("1a2b3c4e5f6g");
        P.synchZoneStart(); 
        break;
      }
      P.synchZoneStart(); 
      y++;
      if(y>2){y=0;}         
    }        

} 

void bigFont(String message){  
    str_len = message.length()+1;                   //length of string
    Serial.println(str_len);
    char char_array_large[str_len];                 //buffer of length of string     
    message.toCharArray(char_array_large, str_len); //copy to buffer   
    strcpy(MesgLARGE, char_array_large);
    createHString(msgH, char_array_large);          //create big font 
    P.displayZoneText(ZONE_UPPER, msgH, PA_CENTER, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_RIGHT, PA_SCROLL_RIGHT);
    P.displayZoneText(ZONE_LOWER, MesgLARGE, PA_CENTER, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_RIGHT, PA_SCROLL_RIGHT); 
}

void smallFont(String message){
    int str_len = message.length()+1;       //length of string
//    Serial.println(str_len);
    char char_array_small[str_len];                 //buffer of length of string
    message.toCharArray(char_array_small, str_len); //copy to buffer   
    strcpy(MesgSMALL, char_array_small);
    P.displayZoneText(ZONE_FREE, MesgSMALL, PA_CENTER, SCROLL_SPEED, PAUSE_TIME, PA_SCROLL_LEFT, PA_SCROLL_LEFT);
}`

MajicDesigns, 2019-01-02:
Your method of creating the high part of the string is flawed. The statements

uint8_t max = 0;
msgH = (char )malloc(sizeof(char)??(max + 2));

allocate a zero sized buffer. They used to allocate a buffer larger than the size of the biggest string. You are then overflowing the buffer. The fact this works sometimes and not others is a matter of luck.

I also really don't get why you need to use Strings to just get the length of a character array in the bigFont() and smallFont(), when you can use the strlen() function to get the length of the character array directly.

This issue has now drifted into a general personal programming support line and I would like you to shift to the Arduino forum if you need that sort of help. There is a Parola thread in the Exhibition/Gallery section that you can use or raise an issue in the Programming section.