MajicDesigns/MD_Parola #38: Static text printing to a zone using displayZoneText?

mrfaptastic, 2019-01-14:
Hello,

I am using Parola on a simple configuration: 8 devices and two Parola zones.

What I want to do is have static text in one zone without any animation. The other zone will have scrolling text.

I can't seem to figure out how to do this. With a single zone (no zone) setup I could simply to Parola.print("XXX"), but this doesn't work reliably if you have more than on zone.

Using "displayZoneText()" forces me to provide an animation and a duration.

Is there any way to simply plonk text into the middle, left or right of a zone without a duration and exist effect. Therefore having text show indefinitely? 

Thank you again for such a great library.

MajicDesigns, 2019-01-14:
Use displayZoneText() with PA_PRINT animation. If you don't want an exit effect (ie, just the same as .print()) then use PA_NO_EFFECT and a delay of 0.

mrfaptastic, 2019-01-14:
Thank you so much! Send you a donation. Keep up the great work.

mrfaptastic, 2019-01-14:
Thank you kindly Marco.

closed 2019-01-14 by MajicDesigns