MajicDesigns/MD_Parola #39: Crash on ESP8266 using parola_zone_mesg example

delhoume, 2019-01-16:
## Crash on ESP8266 (wemos d1 mini) using parola_zone_mesg example

I have a crash on the example for effects 
//  PA_RANDOM, 
//  PA_SLICE,

all I changed is 

#define HARDWARE_TYPE MD_MAX72XX::ICSTATION_HW
#define CLK_PIN   D8 // or SCK
#define DATA_PIN  D7 // or MOSI
#define CS_PIN    D5 // or SS

MD_Parola P = MD_Parola(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);

I am using 8 devices.

## Your Environment
**Library Version:MD_Parola and MD_MAX72XX at version 3.0.2** 
**Arduino IDE version: 1.8.5** 
**Hardware model/type:** 
**OS and Version:MacOS** 

## Steps to Reproduce
Using example provided crashes for two values of effect.
Commenting the two effect gives no crash.

## Expected Behaviour
no crash

## Actual Behaviour
Explain what happens instead. Provide log messages if relevant.

## Code Demonstrating the Issue
````
Insert your code here.
````

delhoume, 2019-01-17:
Thank you for your answer, disabling the software watchdog does not suffice, so I might rewrite the loop. Here is some info on WDs for ESP8266, in case you are not aware of it...

https://techtutorialsx.com/2017/01/21/esp8266-watchdog-functions/

MajicDesigns, 2019-01-30:
Closed due to inactivity from OP.

closed 2019-01-30 by MajicDesigns

thisismyusername123445, 2019-03-13:
i ran into the same problem, solved it by feeding the WDT:

      while (!P.getZoneStatus(2) || !P.getZoneStatus(1) || !P.getZoneStatus(0)){
       P.displayAnimate();
       ESP.wdtFeed();
      }

This solved the problem for me.