MajicDesigns/MD_Parola #40: big fonts justified

wchpikus, 2019-02-24:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Describe your issue here.

## Your Environment
**Library Version:** 
**Arduino IDE version:** 
**Hardware model/type:** 
**OS and Version:** 

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.

## Expected Behaviour
Explain what should happen.

## Actual Behaviour
Explain what happens instead. Provide log messages if relevant.

## Code Demonstrating the Issue
````
Insert your code here.
````

wchpikus, 2019-02-24:
Hello.
Is it possible to adjust text using big fonts different way as left,center,right?
E.g. some pixels of left or right?

MajicDesigns, 2019-02-24:
Your question is unclear:
1. Do you mean one piece of text across the top and bottom rows?
Yes, it is possible BUT you may find that it doe not line up well of you are using the 'folding' arrangement for the modules instead of the 'zig-zag' (see https://arduinoplusplus.wordpress.com/2017/03/15/parola-a-to-z-double-height-displays/). This is because the length of the text may be an value that does not position itself in the same way when you turn it upside down.

2. Do you mean one line of text with some left and some right justified?
You need to use more than 2 zones.

wchpikus, 2019-02-24:
I means, e.g. display text moved four pixels from the left (2 zones, big fonts,synchronize).
I only display text without effects now.
How to move it (adjust)?
I displaying clock and all the time when hour change from e.g.9am to 10, clock move to right ( one more digits appeared).

MajicDesigns, 2019-02-24:
> I only display text without effects now
You always display using an effect, even if it is PA_PRINT.

1. Use a fixed width font for the numbers otherwise the justified font will move the numbers around all the time. The clock example uses this type of font for the time.
2. Place a leading zero if you want a zero in front of the hour, this will automatically align the rest.
3. If you don't want a zero justify to the right (read the library documentation for the right method, I can't remember) or set that justification up in the first place when you use displayText().

Above all, READ THE LIBRARY DOCUMENTATION and _look at the examples_. All the information is in there.

closed 2019-03-13 by MajicDesigns