MajicDesigns/MD_Parola #44: Parola_Zone_TimeMsg.ino esp8266 or esp32

Dancopy, 2019-04-04:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Describe your issue here.

## Your Environment
**Library Version:** 
**Arduino IDE version:** 
**Hardware model/type:** 
**OS and Version:** 

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.

## Expected Behaviour
Explain what should happen.

## Actual Behaviour
Explain what happens instead. Provide log messages if relevant.

## Code Demonstrating the Issue
````
Insert your code here.
````

Dancopy, 2019-04-04:
What should be changed or added in the "Parola_Zone_TimeMsg.ino" sketch to use with esp8266 (only as a board - not WiFi)? Thank you

MajicDesigns, 2019-04-06:
I don't believe that much should be changed from the standard way that the other Arduino boards work. You should probably look at the SPI pin numbers and will need to be careful of the Watchdog timer that will expire if you don't 'feed' it while in loop().

You will need to be more specific if you are having problems, and I would suggest that you ask technical questions on the Arduino forum as there are many other with experience using these libraries on the ESP8266.

closed 2019-04-06 by MajicDesigns