MajicDesigns/MD_Parola #48: Wrong Week's Day

Dancopy, 2019-04-21:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Describe your issue here.

## Your Environment
**Library Version:** 
**Arduino IDE version:** 
**Hardware model/type:** 
**OS and Version:** 

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.

## Expected Behaviour
Explain what should happen.

## Actual Behaviour
Explain what happens instead. Provide log messages if relevant.

## Code Demonstrating the Issue
````
Insert your code here.
````

Dancopy, 2019-04-21:
Greetings Marco!
I'm using your 'Parola_Zone_TimeMsg' sketch;
I loaded a sketch 'DS3231_Serial_Easy' on the DSC3231 RTC to set the time, and when I record the sketch, shows the correct time and date on the display, but Day of the Week is wrong, for example, today (I recorded the sketch) "Sunday, April 21, 2019" is showing "Saturday, April 21, 2019". Would you have any tips for me?
NOTE: On the Serial Monitor shows correct: Sunday, April 21, 2019
Thank you very much

Dancopy, 2019-04-21:
Greetings Marco!
I'm using your 'Parola_Zone_TimeMsg' sketch;
I loaded a sketch 'DS3231_Serial_Easy' on the DSC3231 RTC to set the time, and when I record the sketch, shows the correct time and date on the display, but Day of the Week is wrong, for example, today (I recorded the sketch) "Sunday, April 21, 2019" is showing "Saturday, April 21, 2019". Would you have any tips for me?
NOTE: On Serial Monitor shows correct: Sunday, April 21, 2019
Thank you very much

MajicDesigns, 2019-04-21:
> I loaded a sketch 'DS3231_Serial_Easy' on the DS3231 RTC
I have no idea what this sketch is or what it does.

The DoW on the DS3231 is an arbitrary number cycling between 1 and 7. The meaning of the DoW is in your code (Monday could be 3, it does not matter to the DS3231).
The Parola sketch assigned the label for the day according to the table
`
static const char str[][10] PROGMEM =
  {
    "Sunday", "Monday", "Tuesday", "Wednesday",
    "Thursday", "Friday", "Saturday"
  };
`
In this table Sunday is DoW 1. If you want a different mapping change the table.

closed 2019-04-21 by MajicDesigns

Dancopy, 2019-04-22:
I thought the sketch was his own.
Ok; In the DS3231.cpp file, the table starts on Monday and ends on Sunday and the sketch starts on Sunday and ends on Saturday, so I fixed this in the sketch.
Thank you for the tip.