MajicDesigns/MD_Parola #52: Upper Portion of doublefont Stuck while scrolling .

HarpreetHeera, 2019-07-23:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Describe your issue here.

## Your Environment
**Library Version: MD_Parola-3.1.1** 
**Arduino IDE version: 1.8.9** 
**Hardware model/type: ICSTATION_HW** 
**OS and Version:Windows10** 

## .
I have used example code of MD_Parola Library : Parola_Double_Height_v2

## Expected Behaviour
In double Height font examples: 
Message is Scrolling from Right to left, when scrolling is about to reach an end , the upper portion of double font stuck in 8 LED's modules while lower portion keep moving and finished flawlessly. 
[image]

Thanks

MajicDesigns, 2019-07-23:
- You have a lot of modules (32?). Are you using an external power supply adequately rated for the current? Does it work with fewer modules? Start with 4 top 4 bottom, then add modules. See when it starts to fail. If it works with a few and not with many you have power supply issues.
- Have you checked your wiring to make sure the signals are getting through properly?

HarpreetHeera, 2019-07-24:
Thanks MajicDesigns for assigning the Label.

I am using 2 Zones(Upper Zone and Lower Zone) of length 20 ICSTATION_HW modules. So in total there are 40  ICSTATION_HW modules in series.

Looking forward for your valuable support.

HarpreetHeera, 2019-07-24:
Yes I have checked if I use 16modules (Zone1)+16modules (Zone2) = 32 modules then working perfectly.
Moment i added 4 modules to upper Zones and 4 modules to Lower Zone , making total of 40 modules , it Starts Stucking in last.

closed 2019-08-02 by MajicDesigns

HarpreetHeera, 2019-08-03:
Sorry to say but this issue is still pending. 
I am using external supply.   
My wiring connections are fine. 
All other animations are working. 
I checked that up to the combination of 32 modules it's working. 
But when you add 8 more even single character which is scrolling from right to left get stuck it's upper portion. 

It's not power supply issue. 
I just scrolled single character it's also stuck. 

No solution?

MajicDesigns, 2019-08-03:
If you provide no feedback for 10 days then I assume you fixed the issue.

With all due respect to your expertise, I believe you probably have a power supply issue OR the digital signals need to be boosted or paralleled up. All the signals except for DOUT to DIN (which only travels between single modules) will deteriorate over distance. This is well documented so a simple internet search will tell you what to do.

I am confident the problem is not in the software but in your setup.

HarpreetHeera, 2019-08-04:
I admitted,  I am not an expert in hardware.  That's why I seek valuable support over here.  

Actually as you proposed,  I was working around power supply and tried higher even 5v 3amp Power supply. 

Yes length of digital signal to travel is quite long. 

I think problem is around as you said
"the digital signals need to be boosted or paralleled up. All the signals except for DOUT to DIN (which only travels between single modules) will deteriorate over distance."

You are great.  I didn't think about deterioration of signals. 

I will now do search over the internet and try to fix this. 

I will soon confirm if problem is solved or not. 

Thanks for bearing me. 

Thanks and regards

HarpreetHeera, 2019-08-09:
I have tested voltage level of the Data and CLK signals over the length of 40 modules on oscilloscope. The voltage level available at first module is 4.7V and it degrade to 4.2V on last module. 

I don't know this is enough degradation to stuck upper portion of fonts every time when it scroll from right to left. No other distorted pattern was observed. This error is consistent in nature. 

Also slice animations is running differently at the time of exit in those modules where upper portion of double fonts stuck. Again here pattern of misbehave is consistent. 

I will wait until someone else raise this issue also. Till then I will assume I might have done mistake in writing code or in making hardware connections. 

Thanks

Cyber-One, 2019-12-12:
Just been working through the same problems.
I have 40 modules in two rows of 20 modules.
I am using an external power supply to supply 5.1 V with separate supplies to the upper and lower sets of modules.
I have also run separate lines back to the Arduino for the CS and CLK lines.
Only the Data line which is daisy chained flows through all the modules.
With the full alphabet displayed double height, I get 4.1 V at the ends of both the upper and lower rows. and a total current draw from the PSU of 1.427 Amps.
I have tried different length messages both 26 character and 80 character.
Scrolling is all good until the end of the current message reaches the last 8 modules on the upper row.  At that point the upper row stops and the lower row continues.
At that point the PSU current has dropped to 0.982 Amps.
Testing with a logic analyzer shows no variation in the CS or CLK signals when measured from the end of the upper row. There is still data present the Dout pin, but I haven't broken it down to analyse it yet.

I would have thought, a signal error would have caused problems when the current draw on the system was at its greatest, but this is not the case.

Dropping the PSU voltage down to 4.0V results in a voltage of 2.5V on both the upper and lower rows with the exact same result.
Running the PSU at 6.0V also results in the exact same  result and an end of row voltage 4.3V. PSU current did  increase to 1.77 Amps.

Has any one found a solution to this  problem?

Regards,
Ray

HarpreetHeera, 2019-12-12:
Hi Ray,
I have found only one solution till now and implemented in my project that is just concatenate 25 to 30 empty spaces to the scrolling string before Animate , do it ony when ScrollLeft effect is selected. 
Then it will scroll smooth without any stuck.

Thanks and have a great time.

Regards
Harpreet Singh

Cyber-One, 2019-12-12:
Hello Harpreet,
Thanks for your help.
I found based on your suggestion that 15 tabs work reliably.
This does suggest the hardware and power is fine and that there  is a bug in the software instead.

But at least I will be able to set this up the the silly season :-)

Regards,
Ray