MajicDesigns/MD_Parola #55: Possible SPI conflict on ESP32/8266?

knopserl, 2019-09-30:
I'm using the latest version of the MD_Parola (3.1.1) and MD_MAX3.1.0 using an ESP32 (multiple boards) and Arduino Studio 1.8.10.
I'm using a 4 devices MAX7219 LED Matrix (tried different pieces).
I'm using the default SPI port for the MAX7219 (as there is no option to pass another SPI port).
The LED matrix works fine.
Now I wanted to read a text from an NFC Tag using the popular PN532 RFID Reader which is also connected to the SPI port with another CS Pin.

And now I have the following problem:
Both work seperately but not together. The PN532 Reader is detected but as soon as the text scrolling is used in the loop (using exactly the simple scoll example from MD_Parola), the reader does no longer read/find NFC Tags, if I comment out the  loop part below the reader works.
  if (P.displayAnimate())
  {
    if (newMessageAvailable)
    {
      strcpy(curMessage, newMessage);
      newMessageAvailable = false;
    }
    P.displayReset();
  }

I don't get an error, it does just does not find NFC tags.

I also tried the following:
I used another SPI port (VSPI) with the GPIO Pins 12,13,14,15 which is actually isolated from the default port (HSPI) on 18,19,23,5  it does not work either only if I comment out the P.DisplayAnimate(). So I just assuming that there is a SPI conflict on certain devices.
If I use another RFID Reader (RC522) it seem to work but then I have other problems.

Maybe somebody has an idea what collison might be event if different SPI ports are used.

MajicDesigns, 2019-09-30:
> I'm using the default SPI port for the MAX7219 (as there is no option to pass another SPI port).

This is incorrect. You can use any pins you like. You need to use the alternative method for instantiating the object. Read the documentation or look at the examples. Maybe try this first.

knopserl, 2019-10-01:
Thanks for the quick response. I was refering to the pins but rather to the internal SPI port. The ESP32 has VSPI and HSPI. For both you can reassin the pins but if you don't pass the SPI object, then its just reassigning the pins for defaut SPI which is the HSPI. But it will not help as I already tried to use a seperate SPI port (VSPI) and differnet pins so SPI port/pin separation did not help.

MajicDesigns, 2019-10-01:
No, you did not understand my response. You can use ANY pins, not just hardware SPI. It will bitbash and not be as efficient but it gives you options to try non-SPI object code.

knopserl, 2019-10-01:
I haver alreadytried to use the SW SPI and then it did not work at all (LED stay dark). Do you think it will make a diference if I use any other pins for HW SPI or shall I use HW SPI with just other pin assignments?
As I wrote I used already total different SPI pins and SPI ports for the MX7219/MD_PAROLA and the NFC Reader (PN532) and once the P.Animate is used, the NFC reader stops working.
So what would be your proposal, should I use SW SPI (like below) or HW SPI with just a few other pins for MISO, MOSI and SCK? BTW I don't have so many other pins available as the NFC reader also requires a couple of pins for it's SPI.

I would need to use MD_MXAX72xx  constructor with the reassined pins and pass it to the MD_Parola constructor right?

 MD_MAX72XX(moduleType_t mod, uint8_t dataPin, uint8_t clkPin, uint8_t csPin, uint8_t numDevices=1);

This is what I use and as said, using SW SPI did not work at all.
 
#define MAX_DEVICES 4 // Define the number of displays connected.
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define CLK_PIN   18 // or SCK
#define DATA_PIN  23 // or MOSI
#define CS_PIN    15 // or SS

// HARDWARE SPI
MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);
// SOFTWARE SPI
//MD_Parola P = MD_Parola(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);

MajicDesigns, 2019-10-02:
Just to be really clear, this
MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);
uses the SPI hardware object from the library. This
MD_Parola P = MD_Parola(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);
uses the pins that you deine. You don't appear to be using the second form in the example you provided.

> I would need to use MD_MXAX72xx constructor with the reassined pins and pass it to the MD_Parola constructor right?

No, you give the second form of the constructor (above) the right pins numbers.

closed 2019-10-20 by MajicDesigns