MajicDesigns/MD_Parola #57: ...Scrolling between Time,Weekday & Date with P.print

SurenBono, 2019-11-16:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
..im trying to scroll a fuction P.print text but the library only worked on array

## Your Environment
**Library Version:** =latest
**Arduino IDE version:** = 1 before latest
**Hardware model/type:** =FC16
**OS and Version:**  =XP

## Steps to Reproduce
...check my code below if its workable

## Expected Behaviour
...Smooth scrolling between Time ,Weekdayday and Date

## Actual Behaviour
currently..no animation..just P.print in function call

## Code Demonstrating the Issue
````

// SIMPLE_12H_NTP_LEDMATRIX_v6.1 via WIFIMANAGER Developed by Sroto&Gargees (OpenSourced)
// http://srotogargees.business.site/
// ** Version 6.1 Bugfix **
// Portability  ( connect to any authorised available wifi via a webinterface )
// Secure,Connect to esp.AP with pw,open browser address 192.168.4.1,scanAP(SSID)+ keyin pw with internet connection

#include <FS.h>                     // Included in Arduino IDE / esp8266 setup ..google it
#include <ESP8266WiFi.h>            // http://arduino.esp8266.com/stable/package_esp8266com_index.json
#include <DNSServer.h>              // Included on esp8266 platform setup
#include <ESP8266WebServer.h>       // Included on esp8266 platform setup
#include <WiFiManager.h>            // https://github.com/tzapu/WiFiManager
#include <MD_Parola.h>              // https://github.com/MajicDesigns/MD_Parola
#include <MD_MAX72xx.h>             // https://github.com/MajicDesigns/MD_MAX72XX
#include <SPI.h>                    // Included in Arduino IDE 

const uint16_t SEC_1 = 1000;  

#define HARDWARE_TYPE MD_MAX72XX::FC16_HW

#define MAX_DEVICES 8
#define DATA_PIN  13
#define CS_PIN    15
#define CLK_PIN   14

MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);

const char* ntpServer = "pool.ntp.org";
const long  gmtOffset_sec = 28800;       // My,SG=+8(UTC=GMT=TIMEZONE) *60*60
const int   daylightOffset_sec = 0;

char buffer[80];

void setup()
{
  Serial.begin(115200);
  Serial.println();
  WiFiManager wifiManager;

  IPAddress _ip = IPAddress(10, 0, 1, 78);
  IPAddress _gw = IPAddress(10, 0, 1, 1);
  IPAddress _sn = IPAddress(255, 255, 255, 0);
  
  if (!wifiManager.autoConnect("autoConnect","password")) // scan wifi on smartphone..etc...Open browser 192.168.4.1
  {
    Serial.println("failed, reset reconnect");
    delay(3000);
    ESP.reset();
    delay(5000);
  }
  Serial.println("connected...");
  Serial.println("local ip");
  P.begin();
  P.setIntensity(0);
  P.setTextAlignment(PA_CENTER);
  Serial.println(WiFi.localIP());
  while (WiFi.status() != WL_CONNECTED) {delay(500);}
  P.write("Get NTP");
  delay(SEC_1); 
  configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
  unsigned timeout = 5000;
  unsigned start = millis();
  while (!time(nullptr))
  {
    delay(1000);
  }
  delay(1000);
  WiFi.mode(WIFI_STA);
}

void printLocalTime()
{
  time_t rawtime;
  struct tm * timeinfo;
  time (&rawtime);
  timeinfo = localtime (&rawtime);
  strftime (buffer,80,"%I:%M:%S %p",timeinfo); 
  P.print(buffer);
  delay(SEC_1); 
}

void printDayOnly()
{
  time_t rawtime;
  struct tm * timeinfo;
  time (&rawtime);
  timeinfo = localtime (&rawtime);
  strftime (buffer,80,"%A",timeinfo);
  P.print(buffer);
  delay(SEC_1); 
}

void printDateOnly()
{
  time_t rawtime;
  struct tm * timeinfo;
  time (&rawtime);
  timeinfo = localtime (&rawtime);
  strftime (buffer,80,"%d.%m.%Y",timeinfo);
  P.print(buffer);
  delay(SEC_1); 
}

void loop()
{
   for (int Ti = 0; Ti < 60; Ti++)        // until 25 counts = 25sec
      {printLocalTime();}                 // Display centered 12h time, Minute with Second updates & AM/PM status
      for (int Dy = 0; Dy < 3; Dy++)      // the next each 1 sec until 3 count = 3sec 
      {printDayOnly();}                   // Display full Day name centered ,update Day name if changed inbetween these 3sec
      for (int Dt = 0; Dt < 3; Dt++)      // the next each 1 sec until 3 count = 3 sec 
      {printDateOnly();}    
}

````

MajicDesigns, 2019-11-16:
P.print() does not scroll, it just prints the string to the display. It it does not fit, it it cut off.

You should refer to the examples that come with the library to understand the correct structure for your application. Smooth scrolling requires the use of the P.animate() call and the removal of the delay() calls in your code.

Additional information can be found on the blog site [http://arduinoplusplus.github.com](http://arduinoplusplus.github.com) if you search for "Parola A to Z".

SurenBono, 2019-11-17:
returned 404.....the A-Z to complex/outdated to understand.....could you do simple P.scroll("text"); with header defined scroll speed  on next parola updates..like those LCD control lib

MajicDesigns, 2019-11-17:
Link fixed, my error.

> the A-Z to complex/outdated to understand

On what basis do you say A-Z is outdated? If it is too complex, then perhaps you should find a different library that is easier to use.

> could you do simple P.scroll("text");

P.scroll() is called P.displayScroll(). Have you read the documentation for the library? It is in the docs subfolder of the library folder.

SurenBono, 2019-11-17:
textPosition_t Align = PA_CENTER;
uint8_t scrollSpeed = 5; 
textEffect_t FX = PA_SCROLL_LEFT;

(P.displayAnimate()){P.displayScroll((buffer,Align,Speed,FX); // .....  will this work?

MajicDesigns, 2019-11-17:
No, that is nothing like what it should look like. That will not even compile.

Please refer to the HelloWold example for how to structure and place these two statements.

SurenBono, 2019-11-17:
the HelloWorld missing blinking,set colomn features...etc...even scrolling HelloWorld not present..

EXAMPLE: Scrolling HelloWorld
textEffect_t Fx1= PA_SCROLL_LEFT;
textEffect_t Fx2= PA_SCROLL_RIGHT;
textPosition_t scrollAlign = PA_CENTER;

void loop(void)
{
   if(P.displayAnimate()){P.displayText("HelloWorld",scrollAlign,5,3000,Fx1,Fx2);}
}

MajicDesigns, 2019-11-17:
I don't understand what you have written. Can you please write in complete sentences clearly explaining what your issues are.

SurenBono, 2019-11-21:
..i am trying to recreate a global opensource sourcecode from this unshared youtube scrolling wheather station with clock  https://youtu.be/mUbDF3jzZNM

MajicDesigns, 2019-11-21:
Then the best example to start from is the Zone_TimeMsg example. If you start by understanding how that code works, then add your ESP code to it, you should be in a better position.

SurenBono, 2019-11-21:
Thank you for your kind advice and contribution toward this project..

closed 2020-02-10 by MajicDesigns