MajicDesigns/MD_Parola #58: displayZoneText()/displayText() won't show any effect but PA_NO_EFFECT

deropi, 2019-12-01:
Hi, 
I am trying to get my text scrolling by using displayZoneText() and/or displayText(), but whenever I try to set an effect different from static printing I won't see any text at all. Indeed, I have tried some other effects in addition to scrolling, but they won't show either. 
Here a piece of code for the displayZoneText() try. The commented static text will show just nicely. 
I tried displayText() without setting any zone as well...but did not work..

I would really appreciate some tips!
Thanks in advance
## Code Demonstrating the Issue
````
#include <stdio.h>
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
//#include "Parola_Fonts_data.h"
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 8
#define CLK_PIN   13
#define DATA_PIN  11
#define CS_PIN    10
MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);
void setup()
{
  P.begin(2);
  P.setZone(0,0,3);
  P.setZone(1,4,7);
  P.displayZoneText(0,"smome text", PA_CENTER, 5, 1,PA_SCROLL_LEFT, PA_SCROLL_LEFT );
  //P.displayZoneText(0, "Press Start!", PA_CENTER, 0, 0, PA_PRINT, PA_NO_EFFECT);
  //P.displayZoneText(1, "Press Start!", PA_CENTER, 0, 0, PA_PRINT, PA_NO_EFFECT);
  P.displayAnimate();
}
````

MajicDesigns, 2019-12-03:
displayAnimate() needs to be inside loop() not setup()

MajicDesigns, 2020-02-10:
Closed due to lack of activity from OP

closed 2020-02-10 by MajicDesigns