MajicDesigns/MD_Parola #62: problem with my double height sensor parola.

mokerosan, 2020-01-16:
## Subject of the issue
I am using the library to display the data from the dht22 sensor using "parola_double_height" as an example.

## Your Environment
**Library Version:** 3.3.0
**Arduino IDE version:** 1.8.9
**Hardware model/type:** arduino duemilanove
**OS and Version:** manjaro archlinux

## Steps to Reproduce
I have uploaded the code to the board everything works perfectly for a few minutes until a part of the display turns off, I have already changed the display several times without success.

## Expected Behaviour
that both groups work correctly without the cuts in the display

## Actual Behaviour
at the moment the display of zone 2 works is the one that gives me problems although sometimes it happens to zone 3, but nothing happens to zone 0 and 1.

## Code Demonstrating the Issue
````
// **** INCLUDES *****
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
#include <dht.h>
#include "Font_Data.h"
#include "LowPower.h"
#include <stdio.h>
#include <DS1302.h>
#include <SimpleTimer.h>

dht DHT;
SimpleTimer timer;
#define MAX_MESG  20
#define MAX_MESG2  20
char szMesg[MAX_MESG+1] = "";
char szMesg2[MAX_MESG2+1] = "";
#define DHT22_PIN A5

const int kCePin   = 5;  // Chip Enable
const int kIoPin   = 6;  // Input/Output
const int kSclkPin = 7;  // Serial Clock

// Create a DS1302 object.
DS1302 rtc(kCePin, kIoPin, kSclkPin);
// Define the number of devices we have in the chain and the hardware interface
// NOTE: These pin numbers will probably not work with your hardware and may
// need to be adapted
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 16
#define NUM_ZONES   4

#define CLK_PIN   13
#define DATA_PIN  11
#define CS_PIN    10

MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);

int salida1 = 2;     // verde
int salida2 = 3;    // amarillo
int salida3 = 4;    // naranja
int salida4 = 8;    // rojo
int salida5 = 9;    // violeta

int UVOUT = A0; //Output from the sensor

int REF_3V3 = A1; //3.3V power on the Arduino board

void setup()
{

P.begin(NUM_ZONES);
     P.setIntensity(0);
  P.setZone(0, 0, 3);
  P.setFont(0, BigFontLower);

  P.setZone(1, 4, 7);
  P.setFont(1, BigFontUpper);

  
     
  P.setZone(2, 8, 11);
  P.setFont(2, BigFontLower);

  P.setZone(3, 12, 15);
  P.setFont(3, BigFontUpper);

  pinMode(UVOUT, INPUT);
  pinMode(REF_3V3, INPUT);

 pinMode(salida1, OUTPUT);   // declaramos la variable como salida, OUTPUT
 pinMode(salida2, OUTPUT);   // declaramos la variable como salida, OUTPUT
 pinMode(salida3, OUTPUT);   // declaramos la variable como salida, OUTPUT
 pinMode(salida4, OUTPUT);   // declaramos la variable como salida, OUTPUT
 pinMode(salida5, OUTPUT);   // declaramos la variable como salida, OUTPUT

 digitalWrite(salida1, LOW); // estado inicial del LED apagado
 digitalWrite(salida2, LOW); // estado inicial del LED apagado
 digitalWrite(salida3, LOW); // estado inicial del LED apagado
 digitalWrite(salida4, LOW); // estado inicial del LED apagado
 digitalWrite(salida5, LOW); // estado inicial del LED apagado

     timer.setInterval(5000, sensoruv);
     timer.setInterval(10000, offlight);
     timer.setInterval(30000, sensor);
}

void sensor() {
int chk = DHT.read22(DHT22_PIN);
    
  
  dtostrf(DHT.temperature, 1, 0, szMesg);
  dtostrf(DHT.humidity, 1, 0, szMesg2);   
   P.displayAnimate();

  P.displayZoneText(0, szMesg, PA_CENTER, 0, 0, PA_PRINT, PA_NO_EFFECT);
  P.displayZoneText(1, szMesg, PA_CENTER, 0, 0, PA_PRINT, PA_NO_EFFECT);
  P.displayZoneText(2, szMesg2, PA_CENTER, 0, 0, PA_PRINT, PA_NO_EFFECT);
  P.displayZoneText(3, szMesg2, PA_CENTER, 0, 0, PA_PRINT, PA_NO_EFFECT);
 P.synchZoneStart();
}

void sensoruv() {
  int uvLevel = averageAnalogRead(UVOUT);

  int refLevel = averageAnalogRead(REF_3V3);

  //Use the 3.3V power pin as a reference to get a very accurate output value from sensor

  float outputVoltage = 3.3 / refLevel * uvLevel;

  float uvIntensity = mapfloat(outputVoltage, 0.99, 2.8, 0.0, 15.0); //Convert the voltage to a UV intensity level

  Serial.print("UV Intensity (W/m^2): ");
  Serial.print(uvIntensity);
  Serial.println();

  // Si la lectura del sensor es menor ó igual al valor umbral:
Time t =  rtc.time();
  if (uvIntensity < 2.0 && t.hr >= 7 && t.hr <=21){
                        verde();
                        }
    else if (uvIntensity >= 2.0 && uvIntensity < 2.6 && t.hr >= 7 && t.hr <=21){
                 amarillo(); 
                 }
      else if (uvIntensity >= 2.6 && uvIntensity < 3.2 && t.hr >= 7 && t.hr <=21){
                         naranja();
                   }
           else if (uvIntensity >= 3.2 && uvIntensity < 3.8 && t.hr >= 7 && t.hr <=21){
                                 rojo();
               }
              else if (uvIntensity >= 3.8 && t.hr >= 7 && t.hr <21){
                   violeta();
                   }
  }

void offlight() {
  // Get the current time and date from the chip.
  Time t =  rtc.time();

 if (t.hr < 7 || t.hr >=21){
 digitalWrite(salida1, LOW); // estado inicial del LED apagado
 digitalWrite(salida2, LOW); // estado inicial del LED apagado
 digitalWrite(salida3, LOW); // estado inicial del LED apagado
 digitalWrite(salida4, LOW); // estado inicial del LED apagado
 digitalWrite(salida5, LOW); // estado inicial del LED apagado
                        }

}

void verde() {
 digitalWrite(salida1, HIGH);
 digitalWrite(salida2, LOW);
 digitalWrite(salida3, LOW);
 digitalWrite(salida4, LOW);
 digitalWrite(salida5, LOW);
}

void amarillo() {
 digitalWrite(salida1, LOW);
 digitalWrite(salida2, HIGH);
 digitalWrite(salida3, LOW);
 digitalWrite(salida4, LOW);
 digitalWrite(salida5, LOW);
}

void naranja() {
 digitalWrite(salida1, LOW);
 digitalWrite(salida2, LOW);
 digitalWrite(salida3, HIGH);
 digitalWrite(salida4, LOW);
 digitalWrite(salida5, LOW);
}

void rojo() {
 digitalWrite(salida1, LOW);
 digitalWrite(salida2, LOW);
 digitalWrite(salida3, LOW);
 digitalWrite(salida4, HIGH);
 digitalWrite(salida5, LOW);
}

void violeta() {
 digitalWrite(salida1, LOW);
 digitalWrite(salida2, LOW);
 digitalWrite(salida3, LOW);
 digitalWrite(salida4, LOW);
 digitalWrite(salida5, HIGH);
}

void loop() 
{
   
    timer.run();

}

//Takes an average of readings on a given pin

//Returns the average

int averageAnalogRead(int pinToRead)

{

  byte numberOfReadings = 8;

  unsigned int runningValue = 0; 

  for(int x = 0 ; x < numberOfReadings ; x++)

    runningValue += analogRead(pinToRead);

  runningValue /= numberOfReadings;

  return(runningValue);  

}

float mapfloat(float x, float in_min, float in_max, float out_min, float out_max)

{
  return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
}
````

[image]

MythicalForce, 2020-04-02:
i kinda have the same problem, i have daisy chained 32 MAX7219 and when i upload my sketch it works for a couple of min but then either portions / half / entire display goes black and i have to restart the ESP32 to get it back :S ....

tott, 2020-06-14:
I’m running into the same issue. After a few hours i end up with weird added or missing fragments or changed contrast in some of the modules. Neither displayClear nor displayReset work. 
[image]
[image]

I’m wondering if this could be a degrading CLK signal or something but given that sometimes it also affects the first modules and happens in random patterns i found this unlikely. Would appreciate any guidance for further debugging

MajicDesigns, 2020-06-14:
In MD_Parola.h find
#define STATIC_ZONES 0    ///< Developer testing flag for quickly flipping between static/dynamic zones

and set it to one.

Just below is
#if STATIC_ZONES
#ifndef MAX_ZONES
#define MAX_ZONES 4     ///< Maximum number of zones allowed. Change to allow more or less zones but uses RAM even if not used.
#endif
#endif

Set MAX_ZONES to be a number slightly larger than the number of zones you are using.

See if this makes a difference.

tott, 2020-06-14:
The current master branch does not seem to fit the described code but i went ahead and tried defining MAX_ZONES to one more than my current NUM_ZONES which should have similar effect than what you suggested. I'll report back on my findings.

MajicDesigns, 2020-06-14:
Hmm, yes. That is version 3.3.0 which is not yet released but what I work with. In any case, if there is no switch you have static zones by default and you did the right thing with the change.

tott, 2020-06-15:
Sadly this did not work. Still seeing fragments this morning after implementing the change and leaving it running for 12 hours.

MajicDesigns, 2020-06-15:
It is not clear now many modules/zones you have set up and running.
Also not clear if you are using breadboard or loose connections.
How are the hardware connections? 
Have you tried a pullup on the CS line?
Posting (or sending me) your code may also help me.

tott, 2020-06-16:
Thanks for sticking with me on this. I'll try the pullup on the CS line and if it does not help will prepare a   quick zip with code and some pictures on how it's wired. It's not bread board but soldered connections.  Might need a day or two though until i manage to get results.

tott, 2020-06-17:
After putting a 10k pull-up on the CS of the first module things seem to be running fine for now (>12h). I'll report back if there should be any further issues. Might be worth for other people with similar issues to try this as well and report back.

closed 2020-08-17 by MajicDesigns