MajicDesigns/MD_Parola #64: Cant display scrolling text

Soumojit28, 2020-03-28:
## IMPORTANT
Before submitting this issue
[* ] Have you tried using the latest version of the library?
[ *] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Cant display scrolling text, I need the minimal code to display a text one after one. I made a code but its not working, can you pls help

## Your Environment
**Library Version:** 3.2.0
**Arduino IDE version:** 1.8.10
**Hardware model/type:** Arduino Nano and FS16
**OS and Version:** Windows 10

## Steps to Reproduce
I just connected all things properly, I am using a 4 matrix module FS16 with a arduino nano, The example codes are working fine

## Expected Behaviour
It should scroll the provided text I thing

## Actual Behaviour
Nothing Displays in the matrix

## Code Demonstrating the Issue
````
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>

#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 4

#define CLK_PIN   13
#define DATA_PIN  11
#define CS_PIN    10

MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);

void setup(void)
{
  P.begin();
}

void loop(void)
{
  
  P.displayScroll("Hello", PA_CENTER, PA_SCROLL_LEFT, 100);
  P.displayAnimate();
  P.displayScroll("World", PA_CENTER, PA_SCROLL_LEFT, 100);
  P.displayAnimate();

}
````

MajicDesigns, 2020-03-28:
It takes many many calls to displayAnimate() for the animation to complete. Please read the documentation and look at the structure of the examples to see what is needed. At a minimum you code needs to change to 
`void setup(void)
{
  P.begin();
   P.displayScroll("Hello", PA_CENTER, PA_SCROLL_LEFT, 100);
}

void loop(void)
{
  P.displayAnimate();
}`

closed 2020-03-28 by MajicDesigns

Soumojit28, 2020-03-29:
I tried to understand some example but having some problems, why p.displayScroll in void setup cant I put that in the loop part, and according to your code lets assume I want to print another text after then what should I do?

MajicDesigns, 2020-03-29:
displayAnimate() will return true when the animation is completed. You will see that all the examples have a section similar to this in the main loop

void loop(void)
{
  if (P.displayAnimate())
  {
    // load the new string
    P.displayReset()
  }
}

If you really don't care about doing something else while the string animates then you can also do something like this, although it is not recommended as best practice

// do something to load a new string
while (!P.displayAnimate()) {}  // busy wait loop
// load another string
while (!P.displayAnimate()) {}  // busy wait loop
// etc