MajicDesigns/MD_Parola #66: Different animation speed for InEffect and OutEffect ?

woeperbe, 2020-06-07:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Different animation speed for InEffect and OutEffect ?

I use different animation for in and out  effect and the Martix.PrintText takes the same animation speed for both.
When i use for example Slice for in and Sprite for out the Sprite is too fast ....
Is there a way to define different speeds for in and out animation ?

## Your Environment
**Library Version:** 
3.2.0
**Arduino IDE version:** 
Microsoft Visual Studio and Visual Micro 2020.518.1
**Hardware model/type:** 
NodeMcu V3 ( 8266)
**OS and Version:** 
Windows 10 Version 1909

## Steps to Reproduce

/******************************************************
******* DISPLAY ANIMATED SHORT STRING *****************
*******************************************************
*/
void DisplayString(char * strng) {

	bool ret;
	Matrix.displayText(strng, PA_CENTER, scroll[EEP.scrollmodein].speed * EEP.runspeed, EEP.lookdelay, scroll[EEP.scrollmodein].effect, scroll[EEP.scrollmodeout].effect);
	do {
		ret = (Matrix.displayAnimate());
		ArduinoOTA.handle();
		server.handleClient();
	} while (!ret);

}

MajicDesigns, 2020-06-07:
Not possible at the moment with one function call. The workaround is to split the animation into two parts (in and out) and change the speed between the end of the first part and the start of the second.

This is a good suggestion for a future version, so I will put it on the list.

woeperbe, 2020-06-07:
Thanks for the quick reply
I will try to do the display in two parts for now.

Thank for your great  work on Parola.

Greetings,
Marc Buffet

MajicDesigns, 2020-08-16:
Will be included in 3.5.0 release.

closed 2020-08-16 by MajicDesigns