MajicDesigns/MD_Parola #69: Is it possible to print some text aligned left and other text aligned right on the same screen ?

tavdog, 2020-06-25:
I have tried issuing multiple P.print/P.write command but the last command always overwrites the first.  I tried messing with the P.getGraphicObject()->control(MD_MAX72XX::UPDATE, MD_MAX72XX::OFF); too but that didn't work.

Any hints ? 
Thanks.

MajicDesigns, 2020-06-25:
You need to use 2 separate zone, but his also means that you can no longer just use .print but need to use the animation loop technique (even for a PA_PRINT). There are a few multi-zone examples to follow.

tavdog, 2020-06-25:
When using zones how does that effect drawing pixels directly to the MD_MAX graphics object as in the following
P.getGraphicObject()->setPoint(i, 0, true);   
This type of operation seems to fail when using zones.

MajicDesigns, 2020-06-25:
It shouldn't fail as the MD_MAX72xx library will just set the pixel. What will happen is that the animation will overwrite the entire zone and whatever you have written to it will need to be redrawn. This is explained in the blog article at https://arduinoplusplus.wordpress.com/2018/03/29/parola-a-to-z-mixing-text-and-graphics/

closed 2020-07-19 by MajicDesigns