MajicDesigns/MD_Parola #70: Impossible to connect to the web page 192.168.1.XX with MD_PAROLA Scrolling and an ESP32

Gex31, 2020-06-28:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Describe your issue here.

## Your Environment
**Library Version: MD_PAROLA 3.2.0
**Arduino IDE version: 1.8.13
**Hardware model/type: ESPRESSIF ESP32-WROOM-32 
IDE on Windows 10 

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.
When I test the exemple Simple Wifi Server, it works, I can connect to the web page.

But with MD_PAROLA Scrolling 8266 (I made little changes and rename it in "MD_PAROLA Scrollin ESP32 GEX") i Connect the wifi, but when i try to access the webpage I have a error message in the browser "ERR_EMPTY_RESPONSE" plus in french "Cette page ne fonctionne pas"

I try to change the wifi.h, but nothing change, 

Please can you indicate me the way ?

Thanks a lot

Guillaume

## Expected Behaviour
Explain what should happen.
I want to enter a string on a web page to be displayed on the MAX7219 LED and manage the scrolling effects... That's the purpose of this library I believ..

## Actual Behaviour
Explain what happens instead. Provide log messages if relevant.
The serial moniteur shows we are connect to the Wifi :

>18:38:53.736 -> ⸮⸮Q⸮"uD⸮7L6\⸮",⸮#L⸮h⸮⸮ܤ⸮⸮6H.#⸮i%""u⸮#"⸮f#"⸮:⸮#"⸮f#"⸮f"⸮⸮E⸮⸮?⸮⸮#"ĩEE⸮#⸮"⸮⸮E՜⸮T ⸮rE⸮⸮⸮"⸮#G⸮>u⸮G ⸮E⸮⸮⸮"⸮#⸮"f⸮+4⸮f #⸮c
>18:38:54.573 -> WiFi connected.
>18:38:54.608 -> IP address: 
>18:38:54.608 -> 192.168.43.251

## Code Demonstrating the Issue
_____________________________________________________________________________________
Insert your code here.
// Use the Parola library to scroll text on the display
//
// Demonstrates the use of the scrolling function to display text received
// from the serial interface
//
// User can enter text through a web browser and this will display as a
// scrolling message on the display. Some parameters for the text can also
// be controlled from the web page.
//
// IP address for the ESP8266 is displayed on the scrolling display
// after startup initialisation and connected to the WiFi network.
//
// Connections for ESP8266 hardware SPI are:
// Vcc       3v3     LED matrices seem to work at 3.3V
// GND       GND     GND
// DIN        D7     HSPID or HMOSI
// CS or LD   D8     HSPICS or HCS
// CLK        D5     CLK or HCLK
//
// MD_MAX72XX library can be found at https://github.com/MajicDesigns/MD_MAX72XX
//

//#include <ESP8266WiFi.h>
#include <WiFi.h>
#include <MD_Parola.h>
[... 307 lines cut ...]
   Serial.println(WiFi.localIP());
    
  // Start the server
  server.begin();
  PRINTS("\nServer started");

  // Set up first message as the IP address
  sprintf(curMessage, "%03d:%03d:%03d:%03d", WiFi.localIP()[0], WiFi.localIP()[1], WiFi.localIP()[2], WiFi.localIP()[3]);
  PRINT("\nAssigned IP ", curMessage);
}

void loop()
{
  handleWiFi();

  if (P.displayAnimate())
  {
    if (newMessageAvailable)
    {
      strcpy(curMessage, newMessage);
      newMessageAvailable = false;
    }
    P.displayReset();
  }
}

MajicDesigns, 2020-06-28:
There may be some elements of the JavaScript that gets sent that does not agree with your browser for whatever reason. I would initially try to simplify the message sent to a simple "Hello World" or similar type confirmation that you have loaded the page and your link works properly and then work up the complexity from there.

Gex31, 2020-07-01:
Hello, Thanks for your answer. I success to display a simple text with the serial monitor. I can also change the message with the IP adress to connected to. It works. The DHCP assign an IP adress well, ping is OK. But when I enter the IP adress in my brower (chrome) on my computer or smartphone, same problem : ERR_EMPTY_MESSAGE.

I can also display a web page with the micro controleur with others library, so I think my network is OK (firewall, security filters ok).

That's why I dont understand. I bought other types of ESP maybe is it the firmware/hardware ?

Thanks a lot if you have any idea..

Guillaume

MajicDesigns, 2020-07-01:
The only thing I can think of is to print what is being sent at the ESP32 to the serial monitor to check that it is not an empty message. If you say a plain text message is received, then it is a matter of just trying to incrementally make the message bigger/more complicated to see if it stops working at come point, at which point the last change caused the issue.

Sorry I cannot be more help on this.

closed 2020-07-19 by MajicDesigns

Gex31, 2020-08-02:
Hello, Thanks for your help. It works now with scrolling. But with french caracters like "ç" "é" è" "à" the display is with %^¨ and some things like this. I think it is in the webserver because it's the same on the web form : if I want an "é" it's false on the screen...

Have you any idea where i can select the right font ?

Thank you, Guillaume

MajicDesigns, 2020-08-02:
I refer you to https://arduinoplusplus.wordpress.com/2020/03/21/parola-a-to-z-handling-non-ascii-characters-utf-8/

ve2cfb, 2022-03-06:
Hello Guillaume,
About your 2 Aug 2020 solution for the Web Page not working properly, 
Did you find a solution...?

Thank you, Pierre
@Gex31