MajicDesigns/MD_Parola #71: Parola_Double_Height_v2: last few char "wedge" on upper zone during scroll

supaplextor, 2020-07-18:
See also: https://github.com/supaplextor/arduino-esp-sketches/issues/2

## IMPORTANT
Before submitting this issue
[x] Have you tried using the latest version of the library?
[x] Have you checked this has not already been submitted and/or resolved?
[n/a] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
The last few chars in the message stop scrolling while the lower half finishes.

## Your Environment
**Library Version:** 
3.2.0 and 3.3.0

**Arduino IDE version:** 
1.8.8 (Visual Studio code-insiders dependency)

**Hardware model/type:** 
ESP8266 lolin or clone

**OS and Version:** 
<pre>
supaplex@supaplex-N56JR:~/Arduino$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="18.04.4 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.4 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
</pre>

MajicDesigns, 2020-07-19:
Can you please see if the solution to issue #62 helps resolve this problem?

Also, are you sure the power supply provides enough power for your matrices (24?). You need 5V at a few amps and cannot power the whole display from the Arduino.

supaplextor, 2020-07-19:
I initially tried 5V (last year) with 5V <-> 3.3V level shifters and the 5V rail to power to the LED matrix, but it didn't like it. I checked the current pull of the LED matrix, 350ma peak. Regulator is 800ma which leaves 450ma for the rest of the 3.3v stuff. Right?

MajicDesigns, 2020-07-19:
Each of those modules will be about 220mA if it is fully lit. I think you underestimate the amount of power you need. (see https://arduinoplusplus.wordpress.com/2015/09/12/max7219-and-led-matrix-power-requirements/)

The MAX7219 will work with 3.3V but if you have a long string of matrices and have put the power from one to the other down the line then the actual voltage likely to be too big for the matrices at the end of the daisy chain. Try distributing voltage to both ends of the matrix chain. Same deal for the signals - the actual signal level will drop over three length of the chain.

For that many matrices you really should be using the 5V level shifters with a separate power supply, connected grounds between the 5V and 3.3V supply and pullup/down resistors for the signal lines.

supaplextor, 2020-07-19:
https://raw.githubusercontent.com/supaplextor/arduino-esp-sketches/master/esp8266/Parola_Double_Height_v2/Parola_Double_Height_v2_bb.png - OE is actually tied high.

Right now, it's all dark. I read the datasheet on the txs0108e.pdf and need to fix OE. I have a 5V 5A supply separate from the 3.3v supply (off the USB + regulator on dev board) I have some test leads for my old 10MHZ scope on order.

supaplextor, 2020-07-19:
As to https://github.com/MajicDesigns/MD_Parola/issues/62 his board is 3.3v Atmega? I noticed some 3.3v references in his code.
https://www.arduino.cc/en/Main/arduinoBoardDuemilanove

MajicDesigns, 2020-07-19:
User tott at #62 says he is using and ESP32 board. There are about 3 users in that conversation.

supaplextor, 2020-07-20:
https://github.com/supaplextor/arduino-esp-sketches/blob/master/esp8266/Parola_Double_Height_v2/Parola_Double_Height_v2_bb.pdf updated with better color scheme on wiring. (some colors match the cable feeding the MAX7219 8x8 modules)   Not sure how to wire the right pull-up resistors for the level shifter.

MajicDesigns, 2020-07-20:
Pull up or pull down resistors are there to prevent the signal from floating by providing a 'higher resistance' path to 5V (pull up) or gnd (pull down) when the digital signal is not asserted properly. Once the floating signal is switched on/off it provides a lower resistance than the PU/PD path and the signal changes level.

There are lots of examples of PU/PD resistors on the web if your search, but essentially you connect the digital signal to 5V or GND through a high value resistor (eg, 10k) so that the current passing thru is minimal in parallel to the connection to the input to the MAX7219.

supaplextor, 2020-07-22:
Humm, so 40 of these on 5V will draw 8.8A at full brightness. I have a 90W PC PSU that delivers 10A for 5V. I'm also going to desolder the power bridges between the modules and run individual power for each 4 matrix bundle when my extra wire shows up. Oh, and I'll order the darn pullup resistors too. :)

closed 2020-07-22 by supaplextor