MajicDesigns/MD_Parola #72: How to get text pixel length in order to conditionally set scroll effect

gertst, 2020-08-15:
I programmed a menu system on my led matrix display using a rotary switch.
I want to start a scroll animation of the current menu item when the menu item text is too long to fit the display.
Once a smaller menu item is selected, the scroll should stop.

Is there a way to calculate the required column length for the text to display? If it exceeds the columns of the display I want to enable the scroll effect.
The best I could find is `getTextExtent`, but this is not sufficient for what I need.

Here's what I have so far (part of a class that handles the display):

```
void DotMatrix::loop()
{
  
  if (parola.displayAnimate()) // animates and returns true when an animation is completed
  {
    if (newMessageAvailable)
    {
      parola.displayReset();  
      parola.setTextBuffer(newText.c_str());
      uint16_t colBegin;
      uint16_t colEnd;
      parola.getTextExtent(colEnd, colBegin);
      textLength = colEnd - colBegin;

      //is screen is fully occupied by text? -> assume we need to scroll (BAD IDEA, because the text is not on screen yet...)
      if (textLength >= 31) {
        parola.setTextEffect(PA_SCROLL_LEFT, PA_SCROLL_LEFT); //in and out effect
      } else {
        parola.setTextEffect(PA_PRINT, PA_NO_EFFECT); //in and out effect
      }
      newMessageAvailable = false;
    }
  }
}
```

Any ideas on how to improve this?

Thanks!

MajicDesigns, 2020-08-15:
I have just released v3.4.0 that includes getTextColumns() that does what you need. This was previously a private function and it has been changed to the public class definition. It was done some time ago but the new version not released until now.

closed 2020-08-16 by MajicDesigns

gertst, 2020-08-16:
Thanks Marco, that's exactly what I need :-)
Congrats for your splendid work with this library, btw!

woeperbe, 2020-08-16:
This feature opens nice new  behavior into my project !
Thanks a lot for this masterpiece :-)