MajicDesigns/MD_Parola #73: Starting from version 3.4 Matrix.setIntensity doesn't work anymore ?

woeperbe, 2020-08-17:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Describe your issue here.

## Your Environment
**Library Version: 3.4 and up
**Arduino IDE version: Latest
**Hardware model/type: NodeMcu
**OS and Version:Windows 10 

## Steps to Reproduce
Comiple with 3.4 and up 

## Actual Behaviour
Intensity is always the same independed what value setup in the function 

## Code Demonstrating the Issue
````
void SetLuminance(void) {

	int ch = RealHour(hour());

	if (EEP.brightness[ch] == 0) {
		Matrix.displayShutdown(true);
	};
	if (EEP.brightness[ch] == 1) {
		Matrix.displayShutdown(false);
		Matrix.setIntensity(EEP.lumlow);

		Serial.printf("Intensity set to %d\r\n", EEP.lumlow);
	};
	if (EEP.brightness[ch] == 2) {
		Matrix.displayShutdown(false);
		Matrix.setIntensity(EEP.lumhigh);
		Serial.printf("Intensity set to %d\r\n", EEP.lumhigh);
	};

}
````

MajicDesigns, 2020-08-17:
What values are you actually setting and what is the resistor value being used for the max7219 intensity?

woeperbe, 2020-08-17:
i use FC16 modules and changed nothing compared to using 3.3 where it works normally
I use values 0 and 3 to test the difference

Greetings,
Marc Buffet

woeperbe, 2020-08-17:
I tested again  and again: it works with 3.3 and not with 3.4 and 3.5

MicroSur, 2020-08-17:
Uncomment in MD_Parola.h :
  inline void setIntensity(uint8_t intensity) { _intensity = intensity; _MX->control(_zoneStart, _zoneEnd, MD_MAX72XX::INTENSITY, _intensity); }

woeperbe, 2020-08-17:
This change made it work again !
Thanks for your quick service

MajicDesigns, 2020-08-17:
Yes, for some reason that got commented out. No idea how or why that happened (but I have had a few contributors between versions). Tells me I need to check very carefully from now on as I found a few other things commented out as well. Will be reverted in the next release (3.5.1) after I do a more comprehensive check.

closed 2020-08-17 by MajicDesigns

woeperbe, 2020-08-18:
thanks again for your quick reaction