MajicDesigns/MD_Parola #74: Version 3.5.1 total compile failure

woeperbe, 2020-08-20:
## Subject of the issue
Unable to compile after update to 3.5.1

## Your Environment
**Library Version: 3.5.1
**Arduino IDE version: Latest
**Host OS and Version: Windows 10 / Visual Studio 2007 / Visual Micro 2020.708.7
**CPU Hardware model/type: NodeMcu
**Matrix hardware model/type:** [X ] Parola  [ ] Generic  [ ] ICStation  [X ] FC-16  [ ] Custom (describe how wired)

## Steps to Reproduce
Recompile :-)

## Actual Behaviour
<.....>
MD_Parola.cpp: In member function void MD_Parola::begin(uint8_t)
 MD_Parola.cpp: 63:18: error: call of overloaded 'setSpeed(int, int)' is ambiguous
   setSpeed(10, 10)
MD_Parola.cpp:63: note  candidates are
 MD_Parola.cpp:23: In file included from
MD_Parola.h:1752: note  void MD_Parola  setSpeed(uint16_t, uint16_t)
   inline void setSpeed(uint16_t speedIn, uint16_t speedOut) { for (uint8_t i = 0; i < _numZones; i++) _Z[i].setSpeed(speedIn, speedOut); }
MD_Parola.h:1763: note  void MD_Parola  setSpeed(uint8_t, uint16_t)
   inline void setSpeed(uint8_t z, uint16_t speed) { if (z < _numZones) _Z[z].setSpeed(speed); }
<......>
Error compiling libraries
Build failed for project 'eClock'

MajicDesigns, 2020-08-20:
Compiles here. As a workaround change ie to setSpeed(10) only.

woeperbe, 2020-08-20:
I changed line 63 in MD_parola.cpp and now it compiles !
[image]

MajicDesigns, 2020-08-20:
Re-pushed corrected version

woeperbe, 2020-08-21:
Sorry but i have still the same problem ....
[image]

MajicDesigns, 2020-08-21:
The file on github shows the correct change. How did you get the new version?

woeperbe, 2020-08-21:
i installed the version 3.5.0 and then again 3.5.1 with the Visual Micro Explorer

MajicDesigns, 2020-08-21:
Ok I’ll need to look at what github is doing tomorrow. The change is as per the workaround implemented before.

woeperbe, 2020-08-21:
I downloaded it manually and replaced the library .... it works now !
Github problem ?

busel7, 2020-08-21:
> I changed line 63 in MD_parola.cpp and now it compiles !
> [image]

Thanks for saving my time :+1:

sokigno, 2020-08-21:
The last change on your library was introducing some compatibility problems with already developed software. In my opinion it could be better to renominate setSpeed function for In and Out setup with something more specific (e.g., setSpeedInOut). 

For my code the workaround was to overcast all the integer with uint8_t ( i.e., setSpeed((uint8_t) 0, 10  )

woeperbe, 2020-08-21:
Are you  reading my mind ?

Marc