MajicDesigns/MD_Parola #79: Online Simulations for beautiful MD_Parola examples for Max72xx circuits 😃

Puneethkumarsn, 2021-02-02:
Hi There

first of all, thank a lot for such wonderful working libraries for the Max72xx parts.
We have simulated a few examples here which I could see as a good opportunity for your library users
-to play with the simulations
-to modify, tinker with the code and make their own versions of the project - online
-learn the examples without leaving the browser

Here are the permanent links
https://wokwi.com/arduino/libraries/MD_Parola
* https://wokwi.com/arduino/libraries/MD_Parola/Parola_Ambulance
* https://wokwi.com/arduino/libraries/MD_Parola/Parola_Animation_Catalog
* https://wokwi.com/arduino/libraries/MD_Parola/Parola_CustomChars
* https://wokwi.com/arduino/libraries/MD_Parola/Parola_Fonts
* https://wokwi.com/arduino/libraries/MD_Parola/Parola_HelloWorld
* https://wokwi.com/arduino/libraries/MD_Parola/Parola_Scrolling

We would be happy to provide more attribution if you prefer. all the links to GitHub and the name of the libraries are kept the same.

Can you kindly add it in the ReadMe file?
I will be happy to create a Pull Request. too. let us know 🙂

Best Regards
Puneeth 
https://wokwi.com/

MajicDesigns, 2021-02-02:
Do you mean add a link to the simulation into the READ.ME file for MD_Parola library?

MajicDesigns, 2021-03-09:
No response from OP after 1 month

closed 2021-03-09 by MajicDesigns