MajicDesigns/MD_Parola #82: Can't get ESP8266 to be AP Mode..

CCERocks, 2021-04-23:
Hi MD, im clueless how to change your library example of Parola Scrolling ESP8266 to make it WifiAP Mode.

this is my code. 

`// Use the Parola library to scroll text on the display
//
// Demonstrates receiving and displaying text received from a web interface.
//
// User can enter text through a web browser and this will display as a
// scrolling message on the display. Some parameters for the text can also
// be controlled from the web page.
//
// IP address for the ESP8266 is displayed on the scrolling display
// after startup initialization and connection to the WiFi network.
//
// Connections for ESP8266 hardware SPI are:
// Vcc       3v3     LED matrices seem to work at 3.3V
// GND       GND     GND
// DIN        D7     HSPID or HMOSI
// CS or LD   D8     HSPICS or HCS
// CLK        D5     CLK or HCLK
//
// MD_MAX72XX library can be found at https://github.com/MajicDesigns/MD_MAX72XX
//

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>

// Turn on debug statements to the serial output
#define  DEBUG  0

#if  DEBUG
#define PRINT(s, x) { Serial.print(F(s)); Serial.print(x); }
#define PRINTS(x) Serial.print(F(x))
#define PRINTX(x) Serial.println(x, HEX)
#else
#define PRINT(s, x)
#define PRINTS(x)
#define PRINTX(x)
#endif

// Define the number of devices we have in the chain and the hardware interface
// NOTE: These pin numbers are for ESO8266 hardware SPI and will probably not
// work with your hardware and may need to be adapted
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 8

#define CLK_PIN   D5 // or SCK
#define DATA_PIN  D7 // or MOSI
#define CS_PIN    D8 // or SS

// HARDWARE SPI
MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);
// SOFTWARE SPI
//MD_Parola P = MD_Parola(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);

// WiFi login parameters - network name and password
//const char* ssid = "CABot";
//const char* password = "HelloCoVid";

#ifndef APSSID
#define APSSID "ESPap"
#define APPSK  "thereisnospoon"
#endif

[... 281 lines cut ...]
  Serial.println(myIP);
  server.on("/", handleWiFi);
  server.begin();
  Serial.println("HTTP server started");

  // Set up first message as the IP address
  sprintf(curMessage, "%03d:%03d:%03d:%03d", WiFi.softAPIP()[0], WiFi.softAPIP()[1], WiFi.softAPIP()[2], WiFi.softAPIP()[3]);
  PRINT("\nAssigned IP ", curMessage);
}

void loop()
{
  handleWiFi();
//  server.handleClient();

  if (P.displayAnimate())
  {
    if (newMessageAvailable)
    {
      strcpy(curMessage, newMessage);
      newMessageAvailable = false;
    }
    P.displayReset();
  }
}`

Irfan0209, 2021-04-24:
Sorry, I can't help you yet. I'm still learning too

MajicDesigns, 2021-04-24:
This does not seem to be a bug in the library, rather a request for specific help. Your best bet is to go to the Arduino forum (Arduino.cc) and ask for help there, either in the programming section (https://forum.arduino.cc/c/using-arduino/programming-questions/20) or the Parola support thread (https://forum.arduino.cc/t/led-matrix-display-md_parola-md_max72xx-and-md_maxpanel/167082/1714) where others with experience in what you are doing may be able to help.

closed 2021-04-24 by MajicDesigns

Irfan0209, 2021-04-24:
thanks for the suggestion, i will try later