MajicDesigns/MD_Parola #88: Arduino nano iot 33

eawystr, 2021-07-25:
## IMPORTANT
Before submitting this issue
[x] Have you tried using the latest version of the library?
[x] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Arduino nano iot 33 compatibility

## Your Environment
**Library Version:** 3.5.6
**Arduino IDE version:** 1.8.15
**Host OS and Version:** Windows 10

## Steps to Reproduce
When testing any of the examples with an arduino nano iot 33 it does not work. The display shows horizontal lines going down all the time. I have tested with an arduino nano and everything works fine.

MajicDesigns, 2021-07-25:
I can't see why not and you really have not provided enough details to diagnose anything.

1. The nano iot33 is a 3.3V device and the matrix controller is a 5V device. Do you have a voltage shifter for the signals between the two devices?
2. Is the power supply adequate to power your matrix chain (you don't say how many). You may need a separate power supply from the USB port, with the grounds connected together.

MajicDesigns, 2021-09-18:
Closed due to lack of response

closed 2021-09-18 by MajicDesigns