MajicDesigns/MD_Parola #89: Feature Request ?

woeperbe, 2021-08-07:
I'am using latest version ;-)

Is there a method to set the hardware type in the Parola library just as in the MD_MAX72xx library ?

This wotks perfect in my code for the MD_MAX library but i cannot find the corresponding function call for Parola ?

M.setModuleType(MD_MAX72XX::ICSTATION_HW);

Thanks for your nice work

MajicDesigns, 2021-08-07:
Set it when you create the object. Or you can access the MD_MAX72xx object and use that library’s methods. getGaphicObject() or something like that, from memory.

woeperbe, 2021-08-07:
That works perfect ! You made my day :-)

	Matrix.getGraphicObject()->setModuleType(MD_MAX72XX::ICSTATION_HW);

Thanks again :-) :-)

closed 2021-08-07 by woeperbe