MajicDesigns/MD_Parola #90: displayText, Pause overflow

WonderCRM, 2021-11-11:
Hello,
in MD_Parola.h :: 1459
`  inline void displayText(const char *pText, textPosition_t align, uint16_t speed, uint16_t pause, textEffect_t effectIn, textEffect_t effectOut = PA_NO_EFFECT)`
In method uint16_t pause max 65 seconds, please changing type on uint32_t.

and, how i can set **numDevices** on Setup or Loop, but not in object?
MD_Parola(MD_MAX72XX::moduleType_t mod, uint8_t dataPin, uint8_t clkPin, uint8_t csPin, uint8_t numDevices = 1)

MajicDesigns, 2021-11-11:
Number of devices can only be changed in the constructor. You can create and destroy (new and delete) objects at run time if you need to change the number of modules.

If you feel you need longer delays then please modify the library yourself - you have all the source code.

closed 2021-11-11 by MajicDesigns

WonderCRM, 2021-11-12:
Can you show an example how to create and destroy (new and delete) objects at run time?

WonderCRM, 2021-11-12:
> Number of devices can only be changed in the constructor. You can create and destroy (new and delete) objects at run time if you need to change the number of modules.

Can you show an example how to create and destroy (new and delete) objects at run time?

MajicDesigns, 2021-11-12:
That is just C++. Please refer to a C++ tutorial for this (like this one https://www.programiz.com/cpp-programming/memory-management)