MajicDesigns/MD_Parola #91: Custom Symbols

fruktiz32, 2021-11-28:
## IMPORTANT
Before submitting this issue
[x ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Can not figure out how to add euro and GBP symbols to default font. Should I edit MAX72XX library or Parola library?

## Your Environment
**Library Version:** 
**Arduino IDE version:** 
**Host OS and Version:** 
**CPU Hardware model/type:** 

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.

## Expected Behaviour
Explain what should happen.

## Actual Behaviour
Explain what happens instead. Provide log messages if relevant.

## Code Demonstrating the Issue
````
Insert your compilable code code here (no code snippets).
````

fruktiz32, 2021-11-28:
Figured out... You can add custom characters to standard 128 ASCII table using addChar() function.

closed 2021-11-28 by fruktiz32

MajicDesigns, 2021-11-28:
You can also change the font definition in MD_MAX72xx if you need to change more than just a few characters.