MajicDesigns/MD_Parola #92: Decimal numbers

fruktiz32, 2021-11-29:
## IMPORTANT
Before submitting this issue
[x] Have you tried using the latest version of the library?
[x ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Hello, Thanks for great library. I want to ask about numbers pricision. I find out that if you want to print precise number like 8 numbers or more after decimal it gives you artefacts. It is a way to print any precision numbers? 7 numbers after pointer works OK.

## Your Environment
**Library Version:** 
**Arduino IDE version:** 
**Host OS and Version:** 
**CPU Hardware model/type:** 

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.

## Expected Behaviour
Explain what should happen.

## Actual Behaviour
Explain what happens instead. Provide log messages if relevant.

## Code Demonstrating the Issue
````
Insert your compilable code code here (no code snippets).
sprintf(mtext, "%s", String(number, 8));
      P.displayText(mtext, PA_CENTER,70,0, PA_SCROLL_LEFT, PA_SCROLL_LEFT);

MajicDesigns, 2021-11-29:
The precision is coming from your conversion. Your construct seems strange as is is simply taking a String and copying it to a cstring but using sprintf() to do it.  Seems like there could be better ways to do this.

1. Is mtext large enough to hold the result of the sprintf?
2. Have you tried printing out the String you are converting to make sure it is correct?
3. Have you tried the dstrtof() function to do the conversion?

fruktiz32, 2021-11-29:
I'm using this function with other many displays and it works for decimals perfect but this time does not. Thanks for advice I will try all of them and let you know.

fruktiz32, 2021-11-29:
Looks like dtostrf() worked perfect. Thank you so much!! I've learned something new today, never used this function before. Thanks again!

closed 2021-11-29 by fruktiz32