MajicDesigns/MD_Parola #94: Parallel use with other SPI device on the same bus

MichaelLachmann, 2022-01-14:
I'm using the latest version of the API with an ESP8266.

In parallel I need to read from an ADC which is also connected to the same SPI as the display is. However there are different CS-pins for the display and the ADC.
I do not have enough free pins to use a software simulated spi for any of the devices.

What I've tried was to suspend the display before reading from the ADC and resuming after that. But that does not work.
This is the code I'm using:
```
    bandDisplay.displaySuspend(true);
    SPISettings settings(ADC_CLK, MSBFIRST, SPI_MODE0);
    SPI.beginTransaction(settings);
    int gasValue = adc.read(MCP3202::Channel::SINGLE_0);
    int lightValue = adc.read(MCP3202::Channel::SINGLE_1);
    SPI.endTransaction();
    bandDisplay.displaySuspend(false);
```
but gas and light values are both zwo. If I ommit the whole display code it works, it also works If I do not use the MD_Parola API and only the MD_MAX72xx.

Do you have a suggestion on how to do that correctly?

Thanks,
  Michael

MajicDesigns, 2022-01-14:
The only place SPI comms are done is in the MAX72xx library. Parola uses this library and does nothing on its own and then only if you call animate(), so I am not sure how omitting the Parola portion makes it work differently as you have not shown the code.

Suspending the display stops the updates but I am not sure where it stops all comms to the displays. In any event you seem to have structured ADC SPI transaction as an 'atomic' piece of code, so nothing else should be trying to use the SPI interface while that is going on.

I don't have much suggestion other than to try:
1. Avoid using the 'default' CS pin for display. In some versions of the Arduino libraries the 'default' CS pin is used by the library and you could be getting interference from this.
2. Have you tried pullups/pulldown resistors (can't remember which) for the CS and other SPI lines.
3. With the ESP8266 you will need to have a level converter between the MAX chip and the MCU I/O as the max chip works on 5V. The power supply for the display will also need to be a separate 5V display. Voltage drops could cause weird comms issues.

MajicDesigns, 2022-02-10:
Closed as no response from OP

closed 2022-02-10 by MajicDesigns