MajicDesigns/MD_Parola #95: How to change the text of the display

Pixelherz23, 2022-01-15:
I try to figure out how to display one text after another. 
After Attempt 1 worked flawless, I tried display the words with animations.
Attempt 2 only prints "Hello"
When using attempt 3 the words are mixed sometimes. Like "horld" or "wollo". 
Attempt 3 only prints "hello".
I am new to this library. I looked up  your blog and some other blogs/videos but I didnt found a explanation for changing the display text properly. :/
````
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>

// Define the number of devices we have in the chain and the hardware interface
// NOTE: These pin numbers will probably not work with your hardware and may
// need to be adapted
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 4
#define CS_PIN 5

// Create a new instance of the MD_Parola class with hardware SPI connection:
MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);
// Setup for software SPI:
#define DATA_PIN 23
#define CLK_PIN 18

uint8_t Sspeed = 25;    // default frame delay value
textEffect_t Seffect = PA_SCROLL_LEFT;
textPosition_t Salgin = PA_LEFT;
uint16_t Spause = 2000; // in milliseconds
// Global message buffers shared by Serial and Scrolling functions
#define    BUF_SIZE    75
char curMessage[BUF_SIZE] = { "" };

void setup()
{
  Serial.begin(9600);

  P.begin();
  P.setIntensity(0);
 
}
int i = 0;
void loop()
{
 /*
 //Attempt 3
  if((i%2) >0)
{
    strcpy(curMessage, "hello");
    i++;
}
else
{
    strcpy(curMessage, "world");
    i++;
}
if (P.displayAnimate()){
  P.displayText(curMessage, Salgin, Sspeed, Spause, Seffect, Seffect);
  }

*/

  
  //Attempt 2
  if (P.displayAnimate()){
  P.displayText("Hello", Salgin, Sspeed, Spause, Seffect, Seffect);
  }
  //P.displayReset();
  if (P.displayAnimate()){
  P.displayText("World", Salgin, Sspeed, Spause, Seffect, Seffect);
  }
  
  
 /*
 //Attempt 1
  P.print("Hello");
  delay(5000);
  P.print("World");
  */
}
````

MajicDesigns, 2022-01-16:
Your attempt 1 is ok but it means the processor does not do anything other than print the words.

Attempt 3 is the most correct but you need to increment the index only when the animation has completed
`
if (P.displayAnimate())
{
  if ((i%2) == 0)   strcpy(curMessage, "hello");
  else  strcpy(curMessage, "world");
   i++;
  P.displayText(curMessage, Salgin, Sspeed, Spause, Seffect, Seffect);
}
`
You can see this technique being used in the Parola_Scrolling example.

Pixelherz23, 2022-01-17:
Thanks alot.
I am not sure if I understand it correctly why the changing of `curMessage` and `P.displayText()` has to be in the `if (P.displayAnimate()){}`.
Is it because the `curMessage ` changes too quickly and the display cant keep up with it?

MajicDesigns, 2022-01-17:
displayText() returns true when there is no animation running (ie, no started or just completed). So to change the message you wait until the animation is finished and then change the message and restart the animation.

curMessage is the memory where the message is stored. It is declared in the application and shared with the library to save RAM usage.

If  you change the message while it is animating it will usually work - part of the 'old' message will be on the display and then part of the new message will be displayed after. The problems are when the new message is a lot shorter than the old and the current display point is beyond the end of the new message - the library will unpredictable things.

Please read the library documentation if you don't understand what a class method does. These are extensively documented. In the docs folder of the library, open the file index.html.

closed 2022-01-17 by MajicDesigns

Pixelherz23, 2022-01-18:
Thanks for the help!
I am not used to C++ so much and since there are no tutorials for your library just copy and pastes of your examples, I am quite overwhelmed.
For the record. You probably meant `displayAnimate()` ^^

MajicDesigns, 2022-01-20:
FYI Sopmeone on the Arduino forum had this as a link for a tutorial on MD_Parola
`https://lastminuteengineers.com/max7219-dot-matrix-arduino-tutorial/`